openrazer/openrazer #605: Support Razer Blade 15 (2018) 1532:0233

jbdrthapa, 2018-06-03:
```
Bus 001 Device 005: ID 1532:0233 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0233 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

z3ntu, 2018-06-03:
* Please post the output of `ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

jbdrthapa, 2018-06-03:
```
razer / # ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 220 Jun  3 12:18 .
drwxr-xr-x 4 root root 540 Jun  3 12:18 ..
lrwxrwxrwx 1 root root  10 Jun  3 12:18 usb-11918920-0081IC69_USB_Camera-event-if00 -> ../event14
lrwxrwxrwx 1 root root  10 Jun  3 12:18 usb-Razer_Razer_Blade-event-kbd -> ../event16
lrwxrwxrwx 1 root root  10 Jun  3 12:18 usb-Razer_Razer_Blade-if01-event-kbd -> ../event17
lrwxrwxrwx 1 root root  10 Jun  3 12:18 usb-Razer_Razer_Blade-if02-event-mouse -> ../event18
lrwxrwxrwx 1 root root   9 Jun  3 12:18 usb-Razer_Razer_Blade-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Jun  3 12:18 usb-Razer_Razer_Mamba-event-mouse -> ../event13
lrwxrwxrwx 1 root root  10 Jun  3 12:18 usb-Razer_Razer_Mamba-if01-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10 Jun  3 12:18 usb-Razer_Razer_Mamba-if02-event-kbd -> ../event15
lrwxrwxrwx 1 root root   9 Jun  3 12:18 usb-Razer_Razer_Mamba-mouse -> ../mouse0
```

jbdrthapa, 2018-06-03:
**Product** : Razer Blade 15 Keyboard
**Model**: _(Don't know)_
```
razer / # lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1532:0233 Razer USA, Ltd 
Bus 001 Device 004: ID 1bcf:2c9a Sunplus Innovation Technology Inc. 
Bus 001 Device 003: ID 8087:0025 Intel Corp. 
Bus 001 Device 002: ID 1532:0044 Razer USA, Ltd Mamba Chroma (Wired)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

jbdrthapa, 2018-06-03:
Unfortunately I don't have windows running anymore on my laptop for the wireshark usbmon captures and the screenshots.

jbdrthapa, 2018-06-04:
**pcap file** courtesy **After-Life**

`https://github.com/jbdrthapa/razerblade15/blob/master/razerfiles/blade_usb.pcap`

MikeRamos58, 2018-06-06:
[Blade_15_synapse_screenshots.zip](https://github.com/openrazer/openrazer/files/2074673/Blade_15_synapse_screenshots.zip)

Sysnapse screenshots.

MikeRamos58, 2018-06-06:
2018 Razer Blade 15.6"
1080p 144hz display
gtx1070 max-q
256GB ssd

Model No.
RZ09-02386

Product No.
RZ09-02386E91

captify-daleksandrov, 2018-06-13:
Hello @z3ntu! Did older blade laptops require driver for touchpad to work at least to some extent? As reported on [razer forums](https://insider.razer.com/index.php?threads/razer-blade-15-ubuntu-18-04-trackpad-not-working.35999/) - touchpad is not responsive, although it looks like a HID mouse device. Is this expected behavior or is it something new?

Ahuri3, 2018-06-21:
@jbdrthapa 
Could you try with kernel 18 rc1 ?
There may have been an update concerning the trackpads : https://github.com/torvalds/linux/blob/master/drivers/mfd/intel-lpss-pci.c#L242

jbdrthapa, 2018-06-21:
Installed 4.18_rc1 no change, touchpad still not functioning. Let me know if you want me to check anything else.

MikeRamos58, 2018-06-22:
On a fresh Windows 10 install, you do need to install:
iaLPSS2_i2C_CNL.inf
and
iaLPSS2_GPIO2_CNL.inf
from
http://drivers.razersupport.com//index.php?_m=downloads&_a=viewdownload&downloaditemid=2878&nav=0,350,765,766
to get the touchpad to work.

F4OST, 2018-06-23:
Hi!
Got the same problem, touchpad not working on ArchLinux

gsibley, 2018-06-26:
Hi, same problem here.  2018 Razer Blade 15.6".  Happy to help debug however I can.

z3ntu, 2018-06-26:
Please direct your touchpad problems to https://bugzilla.kernel.org/show_bug.cgi?id=199911 and not here (I'm guessing it's @jbdrthapa who made that bug report - thanks)

montyanderson, 2018-07-12:
Running `4.18.0-041800rc4-generic` after seeing [this tweet](https://twitter.com/Crypt0s/status/1016890540653187072), no avail.

svet-am, 2018-07-12:
I just purchased a Razer Blade 15.6 2018 model.  Mine has the 144Hz 1080p LCD screen.  I read through the thread here but I'm not clear on current status.  I know that the trackpad issue is being tracked in the kernel mailing lists.  What can I do to support/help this effort.  I'd really like to get the Chroma support working.

z3ntu, 2018-07-12:
@svet-am Does that one also have the usb id of 1532:0233 ?

svet-am, 2018-07-12:
Yes, it is.  I'm booted on Windows 10 right now to do some work so I used USBDeview to get the relevant data.  If needed, I can boot into a Live Linux USB stick to fetch the data with lsusb.

==================================================
Device Name       : Razer Blade
Description       : Razer Blade
Device Type       : HID (Human Interface Device)
Connected         : Yes
Safe To Unplug    : Yes
Disabled          : No
USB Hub           : No
Drive Letter      : 
Serial Number     : 
Created Date      : 7/12/2018 9:52:50 AM
Last Plug/Unplug Date: 7/12/2018 9:52:50 AM
VendorID          : 1532
ProductID         : 0233
Firmware Revision : 2.00
USB Class         : 03
USB SubClass      : 00
USB Protocol      : 01
Hub / Port        : 
Computer Name     : [REDACTED]
Vendor Name       : 
Product Name      : 
ParentId Prefix   : 7&30f126ea&0
Service Name      : HidUsb
Service Description: @input.inf,%HID.SvcDesc%;Microsoft HID Class Driver
Driver Filename   : hidusb.sys
Device Class      : 
Device Mfg        : Razer Inc
Friendly Name     : 
Power             : 500 mA
USB Version       : 2.00
Driver Description: Razer Blade
Driver Version    : 10.0.15063.0
Driver InfSection : RzDev_Device.NT
Driver InfPath    : oem22.inf
Instance ID       : USB\VID_1532&PID_0233&MI_01\6&325fb9d4&0&0001
Capabilities      : SurpriseRemovalOK
==================================================

==================================================
Device Name       : Razer Blade
Description       : USB Composite Device
Device Type       : Unknown
Connected         : Yes
Safe To Unplug    : Yes
Disabled          : No
USB Hub           : No
Drive Letter      : 
Serial Number     : 
Created Date      : 7/12/2018 9:52:50 AM
Last Plug/Unplug Date: 7/8/2018 12:33:28 AM
VendorID          : 1532
ProductID         : 0233
Firmware Revision : 2.00
USB Class         : 00
USB SubClass      : 00
USB Protocol      : 00
Hub / Port        : 
Computer Name     : [REDACTED]
Vendor Name       : 
Product Name      : 
ParentId Prefix   : 6&325fb9d4&0
Service Name      : usbccgp
Service Description: @usb.inf,%GenericParent.SvcDesc%;Microsoft USB Generic Parent Driver
Driver Filename   : usbccgp.sys
Device Class      : 
Device Mfg        : (Standard USB Host Controller)
Friendly Name     : 
[... 140 lines cut ...]
ProductID         : 2c9a
Firmware Revision : 0.06
USB Class         : 00
USB SubClass      : 00
USB Protocol      : 00
Hub / Port        : 
Computer Name     : [REDACTED]
Vendor Name       : 
Product Name      : 
ParentId Prefix   : 6&996b5dd&0
Service Name      : usbccgp
Service Description: @usb.inf,%GenericParent.SvcDesc%;Microsoft USB Generic Parent Driver
Driver Filename   : usbccgp.sys
Device Class      : 
Device Mfg        : (Standard USB Host Controller)
Friendly Name     : 
Power             : 500 mA
USB Version       : 2.00
Driver Description: USB Composite Device
Driver Version    : 10.0.17134.1
Driver InfSection : Composite.Dev.NT
Driver InfPath    : usb.inf
Instance ID       : USB\VID_1BCF&PID_2C9A\5&36656249&0&7
Capabilities      : SurpriseRemovalOK
==================================================

montyanderson, 2018-07-12:
Got it working on my machine uses @jbdrthapa's patches.

Wrote some instructions. https://github.com/montyanderson/razerblade15

svet-am, 2018-07-12:
Awesome!  I am traveling right now so I am using Windows for the moment.  I will be back home tomorrow and will certainly try this out.

jbdrthapa, 2018-07-13:
@montyanderson Thanks for creating instructions for getting the touchpad working.

The main intention of this issue creation was to support controlling the lights (chroma) on the Razer Blade 15 2018 keyboard which was working from day 1. We have provided all needed files, captures, screenshots. Please let us know if there is anything else you need for adding the support.

svet-am, 2018-08-02:
@floriansimon1 just to be clear/precise -- you have both trackpad AND openrazer working (for color management, etc) ?

floriansimon1, 2018-08-03:
I haven't tried openrazer yet

z3ntu, 2018-08-03:
@jbdrthapa Unfortunately I can't work with the pcap file you provided because it is not saved as .pcapng. Could you ask whoever made the pcap in the first place to re-do it and save it as pcapng (or anyone else in this thread, I don't really care :wink: )?

cockytrumpet, 2018-08-04:
With @jbdrthapa's patches all of the default effects are working except fire. [Here](https://github.com/cockytrumpet/RazerBlade15) are two captures. One is switching between the default reactive -> fire -> reactive. The other is me trying different color variations

started capture at default fire (top:red, bottom:orange)
null(000000), orange(FF8000)
red(FF0000), yellow(FFFF00)
green(00FF00), blue(0000FF)
cyan(00FFFF), magenta(FF00FF)

I don't know if it's necessary, but the [keymapping is over here.](https://github.com/openrazer/openrazer/compare/master...cockytrumpet:master)

svet-am, 2018-08-09:
FWIW, the build instructions referenced by @montyanderson don't work on Elementary OS.  I tried on both 4.0 and 5.0beta .  The kernel build is failing as-is and I've not had a chance yet to dig into the specifics of why.

svet-am, 2018-08-09:
Have you read through the links?  OpenRazer isn't working either and those instructions fix OpenRazer *in addition to* the touchpad based on the instructions provided.

cockytrumpet, 2018-08-09:
I have and they don't. I understand the confusion since half the people are talking about getting the blinky lights working and the other half came in to talk about a kernel bug effecting the touchpad (on a page for said blinky lights). The instructions above are for patching and compiling a kernel in Ubuntu.
The **kernel** patch for the completely separate issue of the touchpad is in comment 43 in the bugzilla above.

No amount of kernel patching is going to make openrazer implement support for the newest hardware. If you can't wait, you can follow the diffs at https://github.com/jbdrthapa/openrazer to get it up and running unofficially until they do.

svet-am, 2018-08-09:
@cockytrumpet fair enough -- all of that is fair and reasonable.  understood.

alin-timu, 2018-10-03:
Any progress on the support of chroma lightning for the keyboard on Razer Blade 15 ?

svet-am, 2018-10-07:
> Any progress on the support of chroma lightning for the keyboard on Razer Blade 15 ?

I just noticed that in the nightlies of openrazer and polychromatic that my 2018 Razer Blade 15 is now detected and I can customize the keyboard.  BIG CAVEAT -- the keyboard mapping is not currently 100% accurate some some of the keys (like the numeral 1 key) are not working but everything is seems to be okay.

z3ntu, 2018-10-07:
Yes, was fixed with #641

closed 2018-10-07 by z3ntu

herzcthu, 2018-10-08:
openrazer-daemon does not detect driver. I can't use polychromatic or any razer apps.
Here is openrazer-daemon foreground output -
`
sudo openrazer-daemon -F --as-root
The daemon is being run as root.
Starting daemon.
2018-10-09 01:10:33 | razer                          | INFO     | Initialising Daemon (v2.3.1). Pid: 3129
2018-10-09 01:10:33 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerCynosaChroma ----------------------- (1532:022A)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-10-09 01:10:33 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
[... cut ...]
2018-10-09 01:10:33 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2018-10-09 01:10:33 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2018-10-09 01:10:33 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2018-10-09 01:10:33 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2018-10-09 01:10:33 | razer                          | INFO     | Serving DBus

`

herzcthu, 2018-10-08:
Thanks,
I just realized that and tried to change to daily PPA.
After I change to daily PPA and reboot. My keyboard not work completely. I had to reboot to another kernel and remove all openrazer packages. 
Problem is stable version's keyboard razerkbd.so not uninstalled.
Even with apt-get remove --purge not uninstall.
I had to remove/unistall stable kernel module using
dkms uninstall --force openrazer-driver/2.3.1
dkms remove --force openrazer-driver/2.3.1 --all
Then reinstall daily PPA and after reboot, driver works.

loysollivier, 2018-11-19:
Thanks ! That worked for me:
```