openrazer/openrazer #606: Blackwidow Chroma V2, keys not responding.

kierun, 2018-06-14:
I installed the `openrazer-2.3.0` and `polychromatic-0.3.11.1` and plugged in the new keyboard. During bios the keyboard works but as soon as Linux (CentOS7) boots, none of the keys on the key board do anything (press key, nothing happens). The colours on the keyboard work just fine and I am able to switch theme using `polychromatic-controll` just fine.

*Any idea how I can get this to work?*

Here is some debugging information.

```
; uname -a     
Linux serval.isotek.co.uk 3.10.0-862.3.2.el7.x86_64 #1 SMP Mon May 21 23:36:36 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
; dmesg | ag -i razer
[    2.103400] usb 2-1.6: Product: Razer BlackWidow Chroma V2
[    2.103401] usb 2-1.6: Manufacturer: Razer
[    2.104150] input: Razer Razer BlackWidow Chroma V2 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.0/input/input5
[    2.154354] hid-generic 0003:1532:0221.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma V2] on usb-0000:00:1d.0-1.6/input0
[    2.155163] input: Razer Razer BlackWidow Chroma V2 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.1/input/input6
[    2.206258] hid-generic 0003:1532:0221.0004: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma V2] on usb-0000:00:1d.0-1.6/input1
[    2.206914] input: Razer Razer BlackWidow Chroma V2 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.2/input/input7
[    2.207010] hid-generic 0003:1532:0221.0005: input,hidraw4: USB HID v1.11 Mouse [Razer Razer BlackWidow Chroma V2] on usb-0000:00:1d.0-1.6/input2
[    3.644772] razerkbd: loading out-of-tree module taints kernel.
[    3.645663] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
[    3.771283] input: Razer Razer BlackWidow Chroma V2 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.0/input/input11
[    3.778643] input: Razer Razer BlackWidow Chroma V2 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.2/input/input12
[    3.810552] razerkbd 0003:1532:0221.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma V2] on usb-0000:00:1d.0-1.6/input0
[    3.810689] razerkbd 0003:1532:0221.0005: input,hidraw3: USB HID v1.11 Mouse [Razer Razer BlackWidow Chroma V2] on usb-0000:00:1d.0-1.6/input2
[    3.814567] input: Razer Razer BlackWidow Chroma V2 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.1/input/input16
[    3.814733] razerkbd 0003:1532:0221.0004: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma V2] on usb-0000:00:1d.0-1.6/input1
```

[razer.log](https://github.com/openrazer/openrazer/files/2101290/razer.log)

z3ntu, 2018-06-14:
Unfortunately we do not support RHEL based distributions such as CentOS or Scientific Linux.

closed 2018-06-14 by z3ntu

kierun, 2018-06-14:
Well, this sucks… 

Where can I get non official support at all?

z3ntu, 2018-06-14:
https://github.com/moozhub/yum-repo-mooz this is where the packages for these distros are provided. But I don't think that this person can help you...

z3ntu, 2018-06-14:
But it looks like you have secure boot enabled and the razerkbd module is not signed, see https://github.com/openrazer/openrazer/wiki/Troubleshooting#secure-boot

kierun, 2018-06-14:
@z3ntu  Thank you very much indeed. 

I looked at that message however, this is what I get when I type: 

```
; sudo modprobe -v razerkbd                                                     
; echo $?                                                                       
0         
```

and 

```
; openrazer-daemon -Fv
Starting daemon.
2018-06-14 09:19:19 | razer                          | INFO     | Initialising Daemon (v2.3.0). Pid: 21862
2018-06-14 09:19:19 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-06-14 09:19:19 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
[... cut ...]
2018-06-14 09:19:19 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2018-06-14 09:19:19 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2018-06-14 09:19:19 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2018-06-14 09:19:19 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2018-06-14 09:19:19 | razer                          | INFO     | Serving DBus
```

kierun, 2018-06-14:
I am not using "UEFI Secure Boot"… Just checked, I use "legacy" boot.

kierun, 2018-06-14:
Any idea? Or am I left with an expensive brick?… ☹

z3ntu, 2018-06-14:
Try another distro maybe.

kierun, 2018-06-14:
Sadly not possible at work… I got the keyboard since the driver works wonder with my home one on Fedora.

z3ntu, 2018-06-14:
The laptop is also CentOS?