openrazer/openrazer #609: Create conda package for python bindings

keponk, 2018-06-18:
This is probably on the low priority side but currently I use miniconda to manage my python installations on archlinux.
I was getting the " No module named 'openrazer daemon' " error, and found some details on https://github.com/openrazer/openrazer/issues/405

after more digging I realized that the `python-openrazer` package is available on arch's AUR but not on conda main channel (nor conda-forge) and so the only way which I found is very non-advisable was to create a symlink.

This worked to some degree but then ran into python-libs-dependency hell but I figured I'd try it anyway and indeed after lots of symlinking it worked.

Then I was finally able to run `openrazer-daemon -Fv` and now run into a new error:
```
2018-06-18 22:55:59 | razer                          | INFO     | Found device.0: 0003:1532:0224.0004
2018-06-18 22:55:59 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0224.0004/device_type, file is not owned by plugdev
```

That is indeed my Late2016Blade keyboard and my user is part of the `plugdev` group so not sure why this is happening but *I THINK*™ is due to some of the symlinking acrobatics.

I am running archlinux with latest 4.17.2 kernel and tried both `openrazer-meta` and `openrazer-meta-git` from AUR which landed me on the same situation.

z3ntu, 2018-06-19:
Have you tried rebooting?

purplesrl, 2019-03-17:
I know this issue is closed, but there are easier fixes than a reboot:

1. Plug out and plug in the mouse.
2. sudo usbhid-dump -m 1532 

Option two will reset all razer devices, it is useful in case you want to reload and preview the changes, logs from daemon after running command:

```
2019-03-17 11:58:09 | razer                          | INFO     | Serving DBus
2019-03-17 11:58:13 | razer                          | DEBUG    | Device event [unbind]: /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:1532:005E.0017
2019-03-17 11:58:13 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:1532:005E.0017
2019-03-17 11:58:13 | razer                          | WARNING  | Removing 0003:1532:005E.0017
2019-03-17 11:58:13 | razer                          | DEBUG    | Emitted Device Remove Signal
2019-03-17 11:58:13 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:1532:005E.0018
2019-03-17 11:58:14 | razer                          | DEBUG    | Device event [bind]: /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:1532:005E.0018
2019-03-17 11:58:15 | razer                          | INFO     | Found valid device.0: 0003:1532:005E.0018

```