openrazer/openrazer #610: Python Problem With Razer Nostromo on Ubuntu Desktop 16.04

gatewksbury, 2018-06-21:
I am having issues with the daemon.  It works great until I plug in my Razer Nostromo (Model # RZ07-0049).  When I do, the following is shown on terminal while I was debugging:

```
2018-06-21 18:54:13 | razer                          | INFO     | Serving DBus

2018-06-21 18:54:43 | razer                          | DEBUG    | DBus called get_serial_list
2018-06-21 18:54:57 | razer                          | DEBUG    | DBus called get_serial_list
2018-06-21 18:55:02 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:1532:0111.0018
2018-06-21 18:55:02 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1/0003:1532:0111.0019
2018-06-21 18:55:04 | razer                          | INFO     | Found valid device.0: 0003:1532:0111.0019
2018-06-21 18:55:04 | razer.device0                  | INFO     | Initialising device.0 RazerNostromo
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 508, in _collecting_udev_method
    self._add_device(d)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 436, in _add_device
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 86, in __init__
    super(RazerNostromo, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 478, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 441, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 64, in __init__
    self.serial = self.get_serial()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 208, in get_serial
    serial = f.read().strip()
  File "/usr/lib/python3.5/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte

Exception ignored in: <object repr() failed>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 427, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 340, in close
    if not self._is_closed:
AttributeError: 'RazerNostromo' object has no attribute '_is_closed'
```

When I do  `lsusb | grep '1532:'`  , I get the response `Bus 003 Device 012: ID 1532:0111 Razer USA, Ltd` when the device is connected.

z3ntu, 2018-06-23:
Can you please run `cat /sys/bus/hid/drivers/razerkbd/0*/device_serial` and give us the output?

gatewksbury, 2018-06-23:
Right now, with no device plugged in, I get `cat /sys/bus/hid/drivers/razerkbd/0*/device_serialcat: '/sys/bus/hid/drivers/razerkbd/0*/device_serial': No such file or directory`

When I get home, I will plug in the device

z3ntu, 2018-06-24:
You have to plug in the device for that command to work.

gatewksbury, 2018-06-28:
I know @z3ntu , but I wanted to give a baseline.

With device, ` cat /sys/bus/hid/drivers/razerkbd/0*/device_serial` gives the response `����������������������`

z3ntu, 2018-06-28:
Ugh. Gotta love Razer...
Anyways, applying https://github.com/openrazer/openrazer/commit/af84bd6c37b603662fc57f08efd55c71ee1c53a5 to the file `/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/device_base.py` or `/lib/python3/site-packages/openrazer_daemon/hardware/device_base.py`, depending on the distro, should fix that.

gatewksbury, 2018-06-28:
I will try that in the next day or so. 

⁣Sent from TypeApp ​

gatewksbury, 2018-07-02:
Sorry for the delay.  Added suggest from af84bd6 on Ubuntu Desktop 16.04.   The file ended up being found at `/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py`

closed 2018-07-02 by gatewksbury