openrazer/openrazer #612: Missing Abyssus ID

31bbb, 2018-06-28:
Hello,

I just purchased the Razer Abyssus Essential; its ID is: `idVendor=1532, idProduct=006b`

It is not currently recognized; how can I make this work for testing purposes?

z3ntu, 2018-06-28:
* Please post the output of
`lsusb -d 1532:006b -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

31bbb, 2018-06-28:
```
$  lsusb -d 1532:006b -v

Bus 001 Device 008: ID 1532:006b Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x006b 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
$  ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 220 Jun 28 17:40 .
drwxr-xr-x 4 root root 620 Jun 28 17:40 ..
lrwxrwxrwx 1 root root   9 Jun 28 17:40 usb-Razer_Razer_Abyssus_Essential-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Jun 28 17:40 usb-Razer_Razer_Abyssus_Essential-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Jun 28 17:40 usb-Razer_Razer_Abyssus_Essential-if02-event-kbd -> ../event9
lrwxrwxrwx 1 root root   9 Jun 28 17:40 usb-Razer_Razer_Abyssus_Essential-mouse -> ../mouse0
```

Emerald-Creeper, 2019-03-01:
I'm a big fan of your software you've devolved. I've been using OpenRazer since 2017 and it's worked amazingly with my Razer keyboard. But recently, I've purchased a new mouse, The Razer Abyssus essential, and I was wondering if there will be any support for it through polychromatic etc... Thanks again for creating these drivers!

Aratarel, 2019-06-06:
[packets.zip](https://github.com/openrazer/openrazer/files/3261311/packets.zip)
I captured various packets with the tutorial provided. Each packet name is named after each action.

closed 2020-01-04 by hehnope

z3ntu, 2020-01-05:
Let's continue tracking this device at #918