openrazer/openrazer #613: Openrazer renders Ornata Chroma unusable on Gentoo

gentoousername, 2018-07-02:
Recently I made a big mistake: I installed openrazer on my Gentoo. The driver made the keyboard unusable: no ALT-GR, ALT, DELETE, ARROW, KEYS (only on number block), no @, €, on F12 and other F keys there´s the tilde, | not producable because of ALT-GR gone ( the same goes for [ ] { } \ { etc ) and many more issues. I decided to add polychromatic in the hope of regaining control over the keyboard but polychromatic failed to compile [make failed; lessc command not found] . I decided to uninstall the driver and everything it pulled including python-evdev from my system giving the keyboard back to evdev. But that did not help. Typing this on Windows because that´s where it works. My keymap is de-nodeadkeys.  
Any bright idea how I can make my keyboard behave as usual with evdev? What could have gone wrong?

gentoousername, 2018-07-04:
Looks like nobody understands. Somehow the driver must have misconfigured the keyboard. Looking into /dev/input/by-id I see the following symlinks:

`ll
insgesamt 0
drwxr-xr-x 2 root root 180  4. Jul 06:44 .
drwxr-xr-x 4 root root 480  4. Jul 06:42 ..
lrwxrwxrwx 1 root root  10  4. Jul 06:42 usb-Razer_Razer_DeathAdder_Chroma-event-mouse -> ../event14
lrwxrwxrwx 1 root root  10  4. Jul 06:42 usb-Razer_Razer_DeathAdder_Chroma-if01-event-kbd -> ../event15
lrwxrwxrwx 1 root root  10  4. Jul 06:42 usb-Razer_Razer_DeathAdder_Chroma-if02-event-kbd -> ../event16
lrwxrwxrwx 1 root root   9  4. Jul 06:42 usb-Razer_Razer_DeathAdder_Chroma-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9  4. Jul 2018  usb-Razer_Razer_Ornata_Chroma-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9  4. Jul 2018  usb-Razer_Razer_Ornata_Chroma-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9  4. Jul 2018  usb-Razer_Razer_Ornata_Chroma-if02-event-mouse -> ../event4`

As you can see a part of the device Ornata Chroma has been dubbed as mouse. Which is picked up by X .
The above mentioned issues with the keyboard are only there once I start X.

gentoousername, 2018-07-04:
`ls -l /sys/bus/hid/devices/
insgesamt 0
lrwxrwxrwx 1 root root 0  4. Jul 06:49 0003:1532:0043.0004 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/0003:1532:0043.0004
lrwxrwxrwx 1 root root 0  4. Jul 06:49 0003:1532:0043.0005 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.1/0003:1532:0043.0005
lrwxrwxrwx 1 root root 0  4. Jul 06:49 0003:1532:0043.0006 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.2/0003:1532:0043.0006
lrwxrwxrwx 1 root root 0  4. Jul 08:56 0003:1532:021E.0007 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.0/0003:1532:021E.0007
lrwxrwxrwx 1 root root 0  4. Jul 08:56 0003:1532:021E.0008 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.1/0003:1532:021E.0008
lrwxrwxrwx 1 root root 0  4. Jul 08:56 0003:1532:021E.0009 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.2/0003:1532:021E.0009
`
Please help me find out what I see here and how I can get full functionality of my Ornata Chroma back.
`lsusb -v -d 1532:021E

Bus 003 Device 004: ID 1532:021e Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x021e 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
`
Someone amongst you gotta know something!

gentoousername, 2018-07-04:
It seems as if openrazer has mixed up my Ornata keyboard with my Deathadder Chroma mouse, so that vital functions of my keyboard are gone now. So Xorgs evdev driver cannot properly work.
Please, if anyone understands what´ s happened, help me here.

z3ntu, 2018-07-04:
That polychromatic doesn't install is https://github.com/vifino/vifino-overlay/issues/4

If your keyboard doesn't work while openrazer is **not** installed, then there's not much I can do about that. Maybe you broke something in your configuration

gentoousername, 2018-07-04:
Am Wed, 04 Jul 2018 09:40:52 -0700
schrieb Luca Weiss <notifications@github.com>:

> That polychromatic doesn't install is
> https://github.com/vifino/vifino-overlay/issues/4
> 
> If your keyboard doesn't work while openrazer is **not** installed,
> then there's not much I can do about that. Maybe you broke something
> in your configuration
> 
This doesn´ t look right:

xinput --list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Razer Razer DeathAdder Chroma             id=8    [slave  pointer  (2)]
⎜   ↳ Razer Razer DeathAdder Chroma             id=9    [slave  pointer  (2)]
⎜   ↳ Razer Razer Ornata Chroma                 id=12   [slave  pointer  (2)]
⎜   ↳ Razer Razer Ornata Chroma                 id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Razer Razer DeathAdder Chroma             id=10   [slave  keyboard (3)]
    ↳ Razer Razer Ornata Chroma                 id=11   [slave  keyboard (3)]
    ↳ Razer Razer DeathAdder Chroma             id=14   [slave  keyboard (3)]
    ↳ Razer Razer Ornata Chroma                 id=15   [slave
    keyboard (3)]

Openrazer was installed and broke keyboard. After removal with evdev
back the keyboard did not regain its functionality, at least while X is
up. On console, without startx, keyboard works without any issues,
every character where it belongs.

Looks like I have to reinstall my Gentoo from scratch. Or try and get
it back on track booting the system with Sysrescue and in a chroot.

Thank you for your message.

closed 2018-07-06 by z3ntu