openrazer/openrazer #614: [UNKNOWN] Razer Nostromo on Ubuntu Desktop 16.04 Only Controls Brightness

gatewksbury, 2018-07-02:
I want to verify that only the brightness control is changeable for a Razer Nostromo on Ubuntu Desktop 16.04 or is this another issues (maybe related to #610).

_OP NOTE: PLEASE CLOSE AND COMMENT IF NOT A BUG FOR FURTHER INFO!_

[image]

[image]

z3ntu, 2018-07-04:
Is there something else that can be configured on that device (except key remapping - that is not (yet) supported)?

alasta4204, 2018-10-12:
Good morning , I think I am having a similar issue . I have instailed open razer and polycromatic on Ubuntu 16.04 LTS the tray app works and loads polychromatic loads ok and looks like the pic above however the keys are not click  able . Are you supposed to be able to edit key configurations etc ?

Many thanks in advance

lah7, 2018-10-12:
Unfortunately, OpenRazer doesn't support key remapping for gamepads yet. The closest is macro support for the M1-M5 keys on the BlackWidow Chroma.

As mentioned in  https://github.com/openrazer/openrazer/issues/259#issuecomment-296959620, it's still true that changing the lighting is _still_ the only possible option. The daemon needs more work to support this functionality. (Any coders -- pull requests welcome - [some useful info here](https://github.com/openrazer/openrazer/issues/167#issuecomment-274497165))

In Polychromatic, that Nostromo image is just the background which isn't supposed to be interactive.

A fellow user mentioned in https://github.com/polychromatic/polychromatic/issues/80 that they use [Keyboard Master](https://bitbucket.org/conzar/keyboarding-master/src) to reconfigure buttons. I haven't tried it, but it might be the only option in the meantime?

alasta4204, 2018-10-13:
Many thanks for your  prompt reply lah7 . Is there any value in me adding the PPA for the BETAs to get  possible earlier support for the Nostromo?

Warmest regards

lah7, 2018-10-13:
You can if you like, it'll help with testing (once this feature sees progress, but I believe it's stale). Bear in mind that if you wish to use a front-end, it could be a little while before they update to make use of the new features.

alasta4204, 2018-10-13:
Thanks again. I am having some success with Keyboard Master after initially struggling with the sparse documentation . I will keep an eye on this thread.

Warmest regards

closed 2019-03-12 by stale[bot]