openrazer/openrazer #617: Capture Trouble with Razer Huntsman Elite

edualm, 2018-07-07:
Hello,

I'm trying to add support to the Razer Huntsman Elite but I am having a lot of trouble doing so.

I've attached filtered Wireshark captures of me changing the color of the whole keyboard (in static mode) from yellow to red and the color of just a single key (the W key) in Chroma configurator (if it helps, doubt so though).

I can't make sense of the captures though - just changing the static color generates **A LOT** of messages (in the thousands). I've tried replaying some of the messages, but the keyboard does not do anything - stays in the same "default" mode of looping through all colors.

Any help making sense of these logs would be appreciated. I'm not asking for support to be implemented, just for someone pointing me in the right direction so I can do it.

Thanks in advance.

[Change Color of Single Key.pcapng.zip](https://github.com/openrazer/openrazer/files/2173187/Change.Color.of.Single.Key.pcapng.zip)

[Static Yellow to Red.pcapng.zip](https://github.com/openrazer/openrazer/files/2173188/Static.Yellow.to.Red.pcapng.zip)

z3ntu, 2018-07-08:
Unfortunately this device uses "software-based lighting" which isn't yet supported in OpenRazer. You can view the milestone which I've assigned to this issue where you can see other devices using similar techniques to set effects.

I never examined what is the payload of these commands that are sent so I can't tell you what is being sent there unfortunately.
[image]

edualm, 2018-07-08:
@z3ntu which utility did you use for that screenshot? Seems easier than using wireshark to view the logs. Maybe I can try to make some sense of it. Not sure that I'll be able to (I'm not a hardware developer, at all) but I can always try.

terrycain, 2018-07-08:
Those commands just say, here is a whole bunch of RGB values, so essentially if you picked the right start and end packet and replayed them at the same rate, you'd get the same animation

edualm, 2018-07-08:
For anyone also looking into this:

```
char cp[] = {
        0x00, 0x1f, 0x00, 0x00, 0x00, 0x4a, 0x0f, 0x03, 0x00, 0x00, row_number, 0x00, 0x16, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
```

This is the full data part of the message, where the first 10 bytes are always the same, 11th byte is the row number (0x00 -> 0x08 for my keyboard, which also includes the handrest), 12th and 13th bytes also appear to always be the same, and the remaining bytes are R, G, B for each key, in order (apart from the last 2 bytes, which are the checksum and 0x00).

Since my keyboard has 9 rows (6 + another unused one (?) + one for each chroma thing on the keyboard and rest), I need to send 9 USB messages to change the color of the whole keyboard.

terrycain, 2018-07-08:
@edualm 12,13th bytes are start and end column, dont change them, the logic if changed gets annoying.

@z3ntu , driver/daemon should support all that, but set_key_row plus some stuff in the daemon will probably have some defensive coding to limit you on max rows

raptor2101, 2018-07-18:
I'am running a Huntsman on a KDE NEON/Ubuntu. How can i assist you?

jacob-lucas, 2018-09-18:
Also available to assist if needed. Looking forward to seeing this update!

wavetyler, 2018-09-24:
Hello,

What is the latest on support for "software-based lighting"?  Any updates on a path to this?

z3ntu, 2018-09-24:
I won't implement it in openrazer. I have a separate project (currently titled [`razer_test`](https://github.com/z3ntu/razer_test)) that is written (mostly) from scratch and takes care of many shortcomings that exist in openrazer. Once it works properly and has a handful of devices supported, I am planning to replace openrazer with that (so this project will become openrazer-legacy or similar and razer_test will become openrazer).

BobbyBabes, 2018-10-14:
@edualm Would you be so kind to tell what using the Huntsman Elite with Openrazer is like, currently?
What is configurable, and what not ?

P.S.
Or maybe you can tell me what is currently not doable of the list below.
My requirements for a MK are very simple. And most are hardware related.
On keyboard storage for multiple configs. Non-gamer font on keycaps. “~!@#$%^&*()_+” lighted and above or next to the numbers on the row of numbers (not provided by Huntsman, Boooo!). Static per-key RGB lighting. Lighting brightness control. Multimedia keys and volume wheel or knob. Nice to have, but not necessary : macros.
I will be using [EurKEY](https://eurkey.steffen.bruentjen.eu/) for key mapping.

raptor2101, 2018-10-14:
@BobbyBabes: currently nothing is configurable. The MediaKeys are working and the Huntsmann constantly changing the Color of the Lightning...

BobbyBabes, 2018-10-15:
@raptor2101 That’s unfortunate. I need a new keyboard like 2 months ago. Because I spent far too much time finding a MK. I’m going for my second or third choice keyboard. Both of which are fully configurable from the keyboard itself. But not as cool looking as the Huntsman Elite.

untoldwind, 2018-10-18:
I've done some research and found two kinds of commands the synapse software seems to send.

First set-brightness (e.g. 50%):
```
Sending  >>>: status: 0 tx: 1F class: F cmd: 4 data: 3 |0100800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000| crc: 89
Received <<<: status: 2 tx: 1F class: F cmd: 4 data: 3 |0100800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000| crc: 89
```
In razer_test speak that would be a "razer_chroma_extended_matrix_set_brightness" with led_id = 0

I found that "razer_chroma_extended_matrix_get_brightness" with led_id = 0 works as well:
```
Sending  >>>: status: 0 tx: 1F class: F cmd: 84 data: 3 |0100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000| crc: 89
Received <<<: status: 2 tx: 1F class: F cmd: 84 data: 3 |0100800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000| crc: 9
```

Everything else is this "new" 9x22 matrix request mentioned before.
E.g. make everything red:
```
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000000017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AE
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000000017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AE
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000010017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AF
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000010017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AF
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000020017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AC
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000020017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AC
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000030017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AD
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000030017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AD
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000040017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AA
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000040017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AA
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000050017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AB
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000050017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: AB
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000060017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: A8
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000060017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: A8
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000070017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: A9
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000070017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: A9
Sending  >>>: status: 0 tx: 1F class: F cmd: 3 data: 4A |0000080017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: A6
Received <<<: status: 2 tx: 1F class: F cmd: 3 data: 4A |0000080017FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000FF0000000000000000| crc: A6
```

In case your wondering where this output is coming from: I wanted to see if its possible to do this in rust: https://github.com/untoldwind/razer_test_test (Yes we can)

I think I can integrate these findings into razer_test if you could give me some hints.
Especially: It seems like all effects are now done inside this software (i.e. if you set a wave or breath effect, synapse continuously sends the above 9 packages with alternating colors). IMHO this would merit its own RazerDevice class instead of "just" adding more "quirks".

neg-serg, 2018-12-19:
Is there any progress now? I've tried to download and install razer_test_test, but it seems that is does not works at all :(

sumpster, 2019-01-14:
I played around with razer_test from @z3ntu and by luck I discovered that software-based lighting is actually not necessary for the Huntsman Elite. Not sure why Synapse only uses software-based lighting. Maybe that's the way Razer wants to move forward?

The keyboard does support hardware lighting and seems to be more or less compatible with Ornata / Cynosa Chroma. Got all the lighting effects working also in openrazer and send a pull request.

wavetyler, 2019-01-14:
@sumpster Tried to get your branch working on my Huntsman.  The Polychromatic application didn't seem to see it.  Any abnormal steps that need to be taken in order to make this work?

lah7, 2019-01-14:
@wavetyler `razer_test` isn't supported by Polychromatic, since it requires OpenRazer's Python library which isn't ported yet. You can try [RazerGenie's `razer_test` branch](https://github.com/z3ntu/RazerGenie/tree/razer_test) for an alternate GUI though.

sumpster, 2019-01-14:
@wavetyler Just to be sure, I only added the "Huntsman Elite".
The kernel driver works for me via the files in /sys/bus/hid/driver/razerkbd/.
Also the provided python examples such as advanced_effect.py and custom_zones.py work for me.
(https://github.com/openrazer/openrazer/tree/master/examples)

@wavetyler I "ported" my findings to openrazer since all the code was already there. It was just about selecting the right code chunks for this keyboard. I also did add the keyboard in the python daemon, so I would actually expect it to work.

But I am by no means an openrazer expert, so someone might want to have a look at my pull request and check if there is something wrong. (https://github.com/openrazer/openrazer/pull/758)

sumpster, 2019-01-14:
Just tested polychromatic and it works for me. But I think I know why it does not work for you.

I looked around what I could have missed and found something: I did not update install_files/udev/99-razer.rules (which ends up being copied to /lib/udev/rules.d) as I wasn't aware of this mechanism for automatically doing the steps described on https://github.com/openrazer/openrazer/wiki/Setting-up-the-keyboard-driver - I just had no idea how out of date the page really is. :)

The pull request now contains the updated udev rule. Please try if it works for you now. If it doesnt, please check if /lib/udev/rules.d/99-razer.rules contains the text "0226" in line 13. (It did not automatically update this for me.)

wavetyler, 2019-01-14:
Totally works now.  Thanks!!!!

ThunderSon, 2019-03-28:
Hello, I am using Huntsman Elite and the daemon does not state that it has support for the device, despite it being stated in the `supported devices` section. Daemon logs:
```
2019-03-27 21:34:10 | razer                          | INFO     | Initialising Daemon (v2.4.0). Pid: 7421
2019-03-27 21:34:10 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerCynosaChroma ----------------------- (1532:022A)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2019-03-27 21:34:10 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
[... cut ...]
2019-03-28 08:57:09 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2019-03-28 08:57:09 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2019-03-28 08:57:09 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2019-03-28 08:57:09 | razer                          | INFO     | Serving DBus
2019-03-28 08:57:09 | razer                          | DEBUG    | DBus called get_serial_list
```

z3ntu, 2019-03-28:
That's because version 2.4.0 doesn't support the device, and there isn't a version with that device released yet.

closed 2019-03-28 by z3ntu