openrazer/openrazer #619: Package install in Ubuntu 18.04 failing

xnaveira, 2018-07-09:
Hi, trying to install in (K)ubuntu 18.04 via ppa. The installation hangs indefinitely at `Building initial module for 4.15.0-24-generic`, a `ps` reveals he following command running which probably is waiting for an interactive response and thus creating the hang:

> whiptail --backtitle Package configuration --title Configuring Secure Boot --output-fd 11 --nocancel --msgbox Your system has UEFI Secure Boot enabled.  UEFI Secure Boot requires additional configuration to work with third-party drivers.  The system will assist you in configuring UEFI Secure Boot. To permit the use of third-party drivers, a new Machine-Owner Key (MOK) has been generated. This key now needs to be enrolled  in your system's firmware.  To ensure that this change is being made by you as an authorized user, and not by an attacker, you must choose a password now and then confirm the change after reboot using the same  password, in both the "Enroll MOK" and "Change Secure Boot state" menus that will be presented to you when this system reboots.  If you proceed but do not confirm the password upon reboot, Ubuntu will still be able to boot on your system but any hardware that requires third-party drivers to work correctly may not  be usable.

xnaveira, 2018-07-09:
I disabled secure boot and it seems to work. I guess if it only was possible to get the prompt above it could be done "properly" without disabling secure boot.

z3ntu, 2018-07-09:
We don't attempt to create a dialog of this kind, so it's not caused by us ;)
You could try to open a bug report in the Ubuntu bug tracker - I'm guessing the dkms package causes this...

closed 2018-07-09 by z3ntu