openrazer/openrazer #620: keyboard input being recognized as a mouse

moozhub, 2018-07-09:
Hello,

I noticed that there was some lingo in the razerkbd driver source that mentions initializing a device as a [mouse](https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L1694). Would you be able to tell me what this may be doing in relation to [Issue #1](https://github.com/moozhub/yum-repo-mooz/issues/1)? Not sure if the device being recognized as a mouse is what's causing the issue, but the only thing I have going with it at the moment...

`[    3.810552] razerkbd 0003:1532:0221.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma V2] on usb-0000:00:1d.0-1.6/input0`
**`[    3.810689] razerkbd 0003:1532:0221.0005: input,hidraw3: USB HID v1.11 Mouse [Razer Razer BlackWidow Chroma V2] on usb-0000:00:1d.0-1.6/input2`**

Thanks!

terrycain, 2018-07-09:
The keyboard presents itself as 3 usb interfaces, 2 keyboards, and 1 mouse,
the mouse interface is the interface that controls the lighting.

moozhub, 2018-07-10:
Thanks, I figured that was probably the case. I'll close this for now until I have any other questions

closed 2018-07-10 by moozhub