openrazer/openrazer #621: Sort of a pull request to fix some issues with the keyboard driver

kanarok, 2018-07-11:
Hi,

I have a BlackWidow Chroma Tournament Edition and have modified and tested the driver to include 
M1-M5 by pressing KEY_FN + [1-5]

Also I added key_capslock to block that annoying key.
Oh Yeah...added alter_super_fn to user KEY_SUPER as KEY_FN if desired (in Game Mode for example)

I know how to use git, but haven't been able to upload some code for you to review.

z3ntu, 2018-07-11:
You should learn git :wink: https://try.github.io/ the "Learn by doing" things look quite good.

kanarok, 2018-07-11:
Sort of messed up the branch before ;-p

closed 2018-07-11 by kanarok

z3ntu, 2018-07-11:
But you can also just send me the diff and I'll do the git stuff for now

kanarok, 2018-07-31:
I forked and pushed on my own account: https://github.com/kanarok/openrazer
because I did not have write permissions. (Not too new to git but not as experienced as I like to be)

Have fun with it. BTW: Having the Win Key as FN_KEY is killer! I use it all the time

z3ntu, 2018-07-31:
You can use the link https://github.com/openrazer/openrazer/compare/master...kanarok:alter-super-fn to create the pull request :)