openrazer/openrazer #622: Installing 2.3.1 on Mint 17.1 makes Cynosa Chroma not work.

JoshMcCullough, 2018-07-12:
Yesterday I updated openrazer to 2.3.1 as recommended via Mint's update manager. This morning, at work, my Razer Cynosa Chrome keyboard was not working (had power, LEDs and keys illuminated, but no characters when typing). I removed openrazer and the keyboard is now working.

Here is the output from `dmesg` when I plugged the keyboard in with openrazer installed:

```
[ 1572.629142] input: 17:12:18:4C:08:27 as /devices/virtual/input/input36
[ 1655.231188] usb 1-2: new full-speed USB device number 7 using xhci_hcd
[ 1655.380500] usb 1-2: New USB device found, idVendor=1532, idProduct=022a
[ 1655.380506] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1655.380510] usb 1-2: Product: Razer Cynosa Chroma
[ 1655.380514] usb 1-2: Manufacturer: Razer
[ 1655.382208] input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:1532:022A.000E/input/input37
[ 1655.439858] hid-generic 0003:1532:022A.000E: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:00:14.0-2/input0
[ 1655.441401] input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1532:022A.000F/input/input38
[ 1655.499470] hid-generic 0003:1532:022A.000F: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:00:14.0-2/input1
[ 1655.500376] input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:1532:022A.0010/input/input39
[ 1655.559535] hid-generic 0003:1532:022A.0010: input,hidraw6: USB HID v1.11 Mouse [Razer Razer Cynosa Chroma] on usb-0000:00:14.0-2/input2
```

z3ntu, 2018-07-12:
What happens when you run `sudo dkms install openrazer-driver/1.0.0` and `sudo modprobe razerkbd`?

JoshMcCullough, 2018-10-04:
```
josh@josh-stealth-mint:~ $ sudo dkms install openrazer-driver/1.0.0
Error! Could not find module source directory.
Directory: /usr/src/openrazer-driver-1.0.0 does not exist.
```
And `sudo modprobe razerkbd` has no output at all.

z3ntu, 2018-10-04:
Output of `sudo journalctl -b | grep -i razer_mount` ?

JoshMcCullough, 2018-10-04:
No results. However, grepping for `cynosa` ...

```shell
~/.clarakm/page $ sudo journalctl -b | grep -i cynosa
Sep 27 07:32:18 josh-stealth-mint kernel: usb 3-1.3: Product: Razer Cynosa Chroma
Sep 27 07:32:18 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.0/0003:1532:022A.000E/input/input32
Sep 27 07:32:18 josh-stealth-mint kernel: hid-generic 0003:1532:022A.000E: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input0
Sep 27 07:32:18 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.1/0003:1532:022A.000F/input/input33
Sep 27 07:32:18 josh-stealth-mint kernel: hid-generic 0003:1532:022A.000F: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input1
Sep 27 07:32:18 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.2/0003:1532:022A.0010/input/input34
Sep 27 07:32:18 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0010: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input2
Oct 01 08:17:36 josh-stealth-mint kernel: usb 3-1.3: Product: Razer Cynosa Chroma
Oct 01 08:17:36 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.0/0003:1532:022A.0014/input/input36
Oct 01 08:17:36 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0014: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input0
Oct 01 08:17:36 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.1/0003:1532:022A.0015/input/input37
Oct 01 08:17:36 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0015: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input1
Oct 01 08:17:36 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.2/0003:1532:022A.0016/input/input38
Oct 01 08:17:36 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0016: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input2
Oct 01 14:09:34 josh-stealth-mint kernel: usb 3-1.3: Product: Razer Cynosa Chroma
Oct 01 14:09:34 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.0/0003:1532:022A.0019/input/input40
Oct 01 14:09:34 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0019: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input0
Oct 01 14:09:34 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.1/0003:1532:022A.001A/input/input41
Oct 01 14:09:35 josh-stealth-mint kernel: hid-generic 0003:1532:022A.001A: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input1
Oct 01 14:09:35 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.2/0003:1532:022A.001B/input/input42
Oct 01 14:09:35 josh-stealth-mint kernel: hid-generic 0003:1532:022A.001B: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input2
Oct 02 07:48:45 josh-stealth-mint kernel: usb 3-1.3: Product: Razer Cynosa Chroma
Oct 02 07:48:45 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.0/0003:1532:022A.001E/input/input44
Oct 02 07:48:45 josh-stealth-mint kernel: hid-generic 0003:1532:022A.001E: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input0
Oct 02 07:48:45 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.1/0003:1532:022A.001F/input/input45
Oct 02 07:48:46 josh-stealth-mint kernel: hid-generic 0003:1532:022A.001F: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input1
Oct 02 07:48:46 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.2/0003:1532:022A.0020/input/input46
Oct 02 07:48:46 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0020: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input2
Oct 04 06:01:44 josh-stealth-mint kernel: usb 3-1.3: Product: Razer Cynosa Chroma
Oct 04 06:01:44 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.0/0003:1532:022A.0027/input/input51
Oct 04 06:01:44 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0027: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input0
[... cut ...]
Oct 04 06:01:44 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0028: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input1
Oct 04 06:01:44 josh-stealth-mint kernel: input: Razer Razer Cynosa Chroma as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.2/0003:1532:022A.0029/input/input53
Oct 04 06:01:44 josh-stealth-mint kernel: hid-generic 0003:1532:022A.0029: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Cynosa Chroma] on usb-0000:3a:00.0-1.3/input2
```

z3ntu, 2018-10-04:
Even nothing with the journalctl command when you unplug and plug the device back in?

JoshMcCullough, 2018-10-04:
Okay, plugged the keyboard into the laptop _directly_, I use it via a dock/hub daily. Here is the output....

```
~ $ sudo journalctl -b | grep -i razer_mount
Oct 04 12:37:30 josh-stealth-mint razer_mount[22136]: Driver razerkbd
Oct 04 12:37:30 josh-stealth-mint razer_mount[22137]: Driver razerkbd
Oct 04 12:37:30 josh-stealth-mint razer_mount[22139]: Driver razerkbd
Oct 04 12:37:30 josh-stealth-mint razer_mount[22141]: Device_ID 0003:1532:022A.002B
Oct 04 12:37:30 josh-stealth-mint razer_mount[22140]: Device_ID 0003:1532:022A.002D
Oct 04 12:37:30 josh-stealth-mint razer_mount[22142]: Device_ID 0003:1532:022A.002C
Oct 04 12:37:30 josh-stealth-mint razer_mount[22145]: Unbinding 0003:1532:022A.002D from hid-generic
Oct 04 12:37:30 josh-stealth-mint razer_mount[22143]: Unbinding 0003:1532:022A.002B from hid-generic
Oct 04 12:37:30 josh-stealth-mint razer_mount[22146]: Unbinding 0003:1532:022A.002C from hid-generic
Oct 04 12:37:30 josh-stealth-mint razer_mount[22150]: Binding 0003:1532:022A.002B to razerkbd
Oct 04 12:37:30 josh-stealth-mint razer_mount[22151]: Binding 0003:1532:022A.002C to razerkbd
Oct 04 12:37:30 josh-stealth-mint razer_mount[22152]: Binding 0003:1532:022A.002D to razerkbd
Oct 04 12:37:30 josh-stealth-mint razer_mount[22155]: Finished binding 0003:1532:022A.002B
Oct 04 12:37:30 josh-stealth-mint razer_mount[22156]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:022A.002B/
Oct 04 12:37:30 josh-stealth-mint razer_mount[22158]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:022A.002B/
Oct 04 12:37:30 josh-stealth-mint razer_mount[22167]: Finished binding 0003:1532:022A.002C
Oct 04 12:37:30 josh-stealth-mint razer_mount[22168]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:022A.002C/
Oct 04 12:37:30 josh-stealth-mint razer_mount[22170]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:022A.002C/
Oct 04 12:37:30 josh-stealth-mint razer_mount[22172]: Finished binding 0003:1532:022A.002D
Oct 04 12:37:30 josh-stealth-mint razer_mount[22173]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:022A.002D/
Oct 04 12:37:30 josh-stealth-mint razer_mount[22175]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:022A.002D/
```
FWIW I have uninstalled `openrazer-meta` and reinstalled it and the keyboard is working correctly. I re-connected my keyboard through the hub/dock and `journalctl` shows similar activity as before.

Perhaps there is an issue, or some reason why, if I boot the laptop with the keyboard already attached (via the hub/dock), then this output does not show up via `journalctl`?

z3ntu, 2018-10-04:
The udev rule should also trigger when the device is already attached on boot. Not really much that we can do if it doesn't. Sorry...

closed 2018-10-04 by z3ntu

JoshMcCullough, 2018-10-04:
Appreciate you taking the time to diagnose! Thanks.