openrazer/openrazer #623: macOS Support

alexargo, 2018-07-12:
Razer has announced that Synapse 3 will not support macOS. Would it be possible to add Mac support? I'd be willing to work on this if I had a bit of direction. I have a hackintosh that I dual boot between Windows and macOS and would love to be able to use some of my Synapse 3 only devices and Synapse on macOS is pretty crappy to begin with.

spacenono, 2019-07-29:
@alexargo : Did you find a way to use synapse feature on your hackintosh ? (I try to change RGB color of my razer blade 15 2018 on hackintosh mode).

sturze, 2020-08-28:
+1 i also would love to see this on MacOS. 
any news so far?
@spacenono there is a project on GitHub handling Razer RGB stuff on MAC 
https://github.com/1kc/razer-macos

my priority would be DPI on my Deathadder... I don't see any chance to configure this on MacOS. anybody out there with a workaround without synapse 3?