openrazer/openrazer #629: Mamba Tournement Edtion not supported?

Joshua827, 2018-07-23:
I've ran all three of the different programs and all say the daemon doesn't detect a device but linux does. I've gotten razer configuration and it detects it fine. Is it just not running properly yet or am I doing something wrong?

lah7, 2018-07-23:
The mouse is definitely supported, I'm using one right now (ID `1532:0046`). It's likely the OpenRazer driver wasn't installed properly. What's the output of this?

    sudo dkms status openrazer-driver

Which distro are you using? [Check here for some troubleshooting tips](https://github.com/openrazer/openrazer/wiki/Troubleshooting), it could that the kernel headers are not installed.

Joshua827, 2018-07-23:
Ah, I tried to put that code in my terminal and it doesnt do anything, just goes back to where I enter commands. How do I see the distro I am using? Also, I will check the troubleshooting steps and see if I can solve this.

I do thank you for this assistance.

z3ntu, 2018-07-23:
`sudo dkms install openrazer-driver/2.3.1`

Joshua827, 2018-07-24:
Module openrazer-driver/2.3.1 already installed on kernel 4.15.0-29-generic/x86_64
thats the message I got

lah7, 2018-07-24:
Let's see what kernel / distro you're running:

```
uname -a
cat /etc/*release
```

Also, please paste the output of this so we can see what the daemon is reporting:

```
openrazer-daemon -s
openrazer-daemon -Fv
```

Joshua827, 2018-07-25:
Here they are in order requested

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04 LTS"
NAME="Ubuntu"
VERSION="18.04 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

openrazer-daemon -s
Module psutil is missing, not checking for process name
Process 1553 stopped

openrazer-daemon -Fv
Starting daemon.
2018-07-24 20:48:41 | razer                          | INFO     | Initialising Daemon (v2.3.1). Pid: 2590
2018-07-24 20:48:41 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerCynosaChroma ----------------------- (1532:022A)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-07-24 20:48:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
[... cut ...]
2018-07-24 20:48:41 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2018-07-24 20:48:41 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2018-07-24 20:48:41 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2018-07-24 20:48:41 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2018-07-24 20:48:41 | razer                          | INFO     | Serving DBus

It got stuck there

lah7, 2018-07-25:
Seems the output of `uname -a` was missed. That would just confirm the driver DKMS module (`4.15.0-29-generic`) was built for the same kernel (`uname -r`) your system is running.

Try re-installing the driver package and paste the output here:
```
sudo apt install --reinstall openrazer-kernel-modules-dkms
```

Then restart the computer so the driver loads properly.

Joshua827, 2018-07-26:
I did that and it gets stuck at 67% saying 
Unpacking openrazer-kernel-modules-dkms (2.3.1~ubuntu18.04.1) over (2.3.1~ubuntu18.04.1) ...
Setting up openrazer-kernel-modules-dkms (2.3.1~ubuntu18.04.1) ...
Loading new openrazer-driver-2.3.1 DKMS files...
Building for 4.15.0-29-generic
Building initial module for 4.15.0-29-generic

lah7, 2018-07-27:
It will most likely hang there if you've got a UEFI system and secure boot is enabled. Try disabling secure boot from the BIOS/System Setup (by holding F2, F8 or DEL when it boots).

"Secure Boot" is just a system feature preventing unsigned drivers from loading, so it poses a problem when installing this driver. The other option is to [sign the modules yourself](https://github.com/openrazer/openrazer/wiki/Troubleshooting#secure-boot).

Once it's disabled, try running that command again - after a reboot the mouse should be detected by the daemon/applications. :+1:

Joshua827, 2018-07-28:
So I finally got it, I got this message at the end does it mean anything?

ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'

z3ntu, 2018-07-29:
Yes. Read the message.

closed 2018-11-27 by z3ntu