openrazer/openrazer #630: Information request

2E0PGS, 2018-07-24:
Greetings Devs,

I have a Razor DeathAdder Chroma.

I am trying to figure out what features of the keyboard require this daemon or the synapse software. 

If any of you devs can help me understand what gets stored in the keyboards EEPROM and what is purely held in the software I would be greatful.

For example multiple lighting profiles are stored in the software and changing between those requires the software running but one of the default lighting mode can be selected and the keyboard will know how to run that from its own memory.

Another example is macros all require the software to be running for them to be recorded or played back but are those macros held in software or in memory on the keyboard. From my tests is the prior.

Any other diagrams, information or architecture info you can tell me to fill me in would be greatly appreciated. 

Here is a repo im documenting this info in for public knowledge since I cannot find this information documented in one place elsewhere. Most of this im having to figure out myself by testing: https://github.com/2E0PGS/razor-deathadder-chroma-features

This information is also useful for understand other Razor products since they will be likely similair.

2E0PGS, 2018-07-24:
`sync_effects_enabled` is a config option this must mean the keyboard will sync it's choosen effect back the host to keep it updated if you move between computers. So that would prove the keyboard stores atleast some effects info. How much however I still dont know. Would custom effects be stored on the keyboard.

z3ntu, 2018-07-24:
1. A razor is what you use to trim your beard. Razer is a gaming hardware manufacturing company.

2. For what the devices save themselves, grep for 'VARSTORE' in the driver/ directory and where this is passed, the device will (probably) remember the effect even if it is unplugged and plugged into another device.

3. The sync effects option is that when you e.g. select the Spectrum effect for your mouse that it also applies that effect on the keyboard. The logic for that is in some python file in the daemon.

4. For the rest: no idea

2E0PGS, 2018-07-25:
Hah yes 1. is correct however I wont be shaving my beard anytime soon :P I corrected the spelling on that repo now. Oh good Github will automagically redirect from that old link to new.

Ok thanks I will have a look and see what I find while grepping to see what gets stored on what devices hopefully.

I see I saw mention of some feature request ref synced devices lighting that makes sense now.

Ok thanks for the input so far hopefully others can chime in to fill a few blanks.

sevu, 2018-09-30:
Interesting might as well be that part from https://support.razer.com/software/synapse-3:
> Hybrid storage is a new feature that is being introduced with Razer Synapse 3. It gives users the ability to store a select number of profiles from the cloud and onto the device's on-board memory. Hybrid storage is available only on select devices that support the feature.

terrycain, 2018-09-30:
The use of eeprom was discontinued years ago. If they start again and I get hold of a keyboard that uses the hybrid storage I'll look into it

closed 2019-02-28 by stale[bot]