openrazer/openrazer #631: Documentation request

2E0PGS, 2018-07-24:
It would be great if there was documentation detailing the current intergrated feature list of stable and develop builds.

That would save me guessing what I should expect to be exposed by the library to other programs.

z3ntu, 2018-07-24:
I don't really understand what you mean... The list of supported devices is in the README.md file (master & stable depending on what you want) but I don't think this is what you seek.

2E0PGS, 2018-07-25:
So yeah that shows supported devices but I think there should be a documented list of features openrazor supports perhaps broken down by device. 

So for example does openrazor support custom colour modes in stable? Does it support custom colours in develop? Do all keyboard that are "supported" also work with that feature? 

Are macros still "on the fly" or is there a facility to rebind keys in stable? If not is there such facility in develop?

Do you see what I mean now? Like a feature list not a supported device list.

Even at the most basic it would look something like:

# Stable

* Custom colours
* Macros (On the fly only)
* Set DPI

# Develop

* Rebinding of macro keys

etc

2E0PGS, 2018-07-25:
Perhaps a roadmap style is another option.

z3ntu, 2018-07-25:
The contents of master get pushed into stable at every release which happened about every 40 commits or so in the past (or when there's something worth releasing). And as we are not a nodejs project with 10 commits per day not much changes over the course of months ;)

I don't have any new features planned except adding the occasional device support, so we don't really have a great roadmap currently

And some features e.g. macro support are quite limited so I wouldn't declare them as big feature or anything.

2E0PGS, 2018-07-25:
Right but say im not aware of half the features currently in the project/library could you alteast list some of those now?

terrycain, 2018-07-25:
This project actually does very little, it acts as an interface between the raw hardware and client applications like RazerGenie and Polychromatic. As for what the interface can do, generally the equivelent of the lighting and misc configuration that Razer Synapse can do. It can change the led effects of your devices and change DPI on mice to name a few.

2E0PGS, 2018-07-25:
To name a few but what are the rest? That's what im trying to find out as it's not documented. Does it support custom colour modes aswell as built in modes.

This should be a list of stuff laying out what the interface/driver can and cant do compared to the hardware functions available.

2E0PGS, 2018-07-25:
Is key rebinding possible? Example: Can I remap my macro key to be another keyboard key like F12?

z3ntu, 2018-07-25:
Your example should be possible to accomplish using razerCommander. But in general no, key rebinding is not supported.

> This should be a list of stuff laying out what the interface/driver can and cant do compared to the hardware functions available.

What the hardware supports is supported by OpenRazer

> Does it support custom colour modes aswell as built in modes.

If I understand the question correctly, the only effect that's currently implemented in software is the ripple effect, all others are hardware effects.

2E0PGS, 2018-07-25:
Thank you that has cleared alot of my questions up.

Yes you have understood that last question correctly.

> What the hardware supports is supported by OpenRazer

I don't this this does list what hardware features are implemented per a device. All I see is what devices are supported but no mention if the interface supports interacting with all its hardware features. Is it assumed if the device is supported then its fully supported with all hardware features available via the library. If so then key rebinding should then be possible for all devices with such capability.

z3ntu, 2018-07-25:
> Is it assumed if the device is supported then its fully supported with all hardware features available via the library.

Yes, that is the case.

> If so then key rebinding should then be possible for all devices with such capability.

No, because key rebinding is not a hardware effect or anything. And it's not supported for any device..

closed 2018-07-25 by z3ntu

2E0PGS, 2018-07-25:
Are recorded macros saved on the hardware device or on the computer via software?

z3ntu, 2018-07-25:
Software. If you exactly want to know how look I into the source code of razerCommander

2E0PGS, 2018-07-25:
Thanks I will look into that.

Yes I do want to know exactly so I can document it for my personal research. As per the info I was requesting in issue #630