openrazer/openrazer #632: Razer Base Station Chroma

edualm, 2018-07-24:
Apparently the Base Station Chroma isn't supported by the driver, so I thought I'd share my findings.

This device is weird - I wasn't being able to change the color settings by replaying the USB packets. Apparently, it needs some "magic packets" to be sent before it starts accepting chroma settings, for some reason.

Apart from that, it seems to work like any other synapse 3 / software chroma device.

I don't use Linux but macOS, so I coded a driver for it based on code from this repository, and I'd like to give back to this project though, even though I'm not a Linux developer.

Relevant code is here: https://github.com/FruityRazer/FruityRazer/blob/master/FruityRazer/Core/RazerSoftwareChroma.c#L106 (line 106 up to the end of the file). 

`razer_base_station_pre` sends those USB packets that enable chroma.

There's also some code to support the Huntsman Elite in that file.

Let me know if I can be of assistance in anything. 😄

terrycain, 2018-07-24:
> master/FruityRazer/Core/RazerSoftwareChroma.c#L106 (line 106 up to the

Try putting the device into "driver" mode first then sending the rgb frames
will work.

kells64000, 2019-10-12:
i have the same problem.
how can i help for this issue ?

redaxmedia, 2020-03-12:
@z3ntu What is the state of this issue? Could you merge https://github.com/lah7/openrazer/tree/base_station_chroma or do should I provide a pcap?

lah7, 2020-03-12:
I have a pull request ready for this device, but #1003 needs to be merged first.

closed 2020-03-14 by z3ntu

redaxmedia, 2020-03-15:
Can this be used in the daily package soon?

z3ntu, 2020-03-15:
It should already be included in the latest daily package from the PPA. On Arch Linux you can just build openrazer-git to get these changes.

redaxmedia, 2020-03-15:
Well, openrazer-meta is on version `2.7.0-1~478~ubuntu19.04.1` but seems not to be supported.

z3ntu, 2020-03-15:
If other devices work fine with openrazer, I'm guessing you either haven't rebooted (i.e. loaded the new kernel modules yet) or there's some bug in the loading (which I doubt)

redaxmedia, 2020-03-15:
It was the reboot... :-)

Thanks

harrisonthorne, 2021-03-18:
I recently got a Base Station Chroma V2, but it looks like only the V1 is supported as of now. Should we reopen this issue or should I open a new one?

lah7, 2021-03-18:
@harrisonthorne Please do create a new issue and follow the template. If you're lucky and they're similar, implementation might be a replica (with tweaks) to https://github.com/openrazer/openrazer/pull/1043/files

PR #1043 (2020-03-14): Add support for Razer Base Station Chroma
Also known as the Razer Headphone/Headset Stand. The effects, brightness and custom matrix works as expected. The product is similar to the HDK (#1003)

Closes #632.