openrazer/openrazer #637: Import issue in openrazer-daemon

muesli, 2018-08-09:
Running Arch, having `openrazer-daemon` 2.3.1-2 from AUR installed, I get this error when trying to start it:

```
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 11, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
ModuleNotFoundError: No module named 'openrazer_daemon'
```

z3ntu, 2018-08-09:
Did you rebuild the openrazer packages after the latest python update (3.6.something -> 3.7.something)? You can get a list what you have to rebuild with `pacman -Qo /usr/lib/python3.6/site-packages/`

muesli, 2018-08-09:
Thanks, that solved it! I wasn't aware manual interaction is required in such a situation. Maybe that's something for the FAQ until official packages have been adopted widely?

closed 2018-08-09 by muesli

z3ntu, 2018-08-09:
Well this is one of the reason you should know what you are doing when you are using the AUR. :)

muesli, 2018-08-09:
Fair enough, but if that means you need to know the ins & outs of all the involved languages, dependencies and components, I pretty much doubt anyone's qualified enough to use the AUR ;-)

Since Razer's official page is linking to the AUR, I'd still appreciate a note / remark there.

z3ntu, 2018-08-09:
The only thing you need to know for python AUR packages is exactly that path change with all major python versions. Perl packages also need something similar. And for normal C++ stuff you'll notice when some application says that it can't find some .so file, then you gotta rebuild.

What official page from Razer..? Could you link that for me?