openrazer/openrazer #638: Support for Mageia Linux

stroibe974, 2018-08-10:
Hello!
Would you please consider adding support for Mageia linux distribution?
Thank you very much!

egoitzro, 2018-09-16:
I'm also interested in having drivers for mageia. Thank you

z3ntu, 2018-09-16:
You can probably use the spec found at https://github.com/openrazer/OBS-packaging/blob/master/openrazer.spec then.

z3ntu, 2018-09-16:
Packages now exist in the OBS repo (follow the Fedora/openSUSE/Debian instructions on the website and select Mageia on OBS site) but it currently doesn't work because of (at least) two issues:
Nr. 1: https://bugs.mageia.org/show_bug.cgi?id=23589 - this hinders the systemd user service to start (but can be worked around by manually launching openrazer-daemon)

Nr. 2 : The dkms version is really old and fails while running `dkms install -m openrazer-driver -v 2.3.1` with the error
```
Error! DKMS tree does not contain: openrazer-driver-2.3.1
Build cannot continue without the proper tree.
```
Even though the source files exist at the correct location. This should be fixed by an upgrade of dkms in Mageia.

z3ntu, 2018-09-17:
Packages do work now. 
Go to https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta and follow the instructions for Mageia.
The only thing that doesn't work as it is supposed to do, is running `systemctl --user start openrazer-daemon.service` but you can start the daemon by just accessing the D-Bus interface, e.g. via a frontend (polychromatic and RazerGenie are packaged and live in the same repo).

closed 2018-09-17 by z3ntu