openrazer/openrazer #639: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11

Mathux, 2018-08-11:
Hi,
I am currently creating a systemd daemon to be able to do a keyboard startup animation. I created a python script where I use DeviceManager from openrazer.client.
This script works well after I am in my session, but I am trying do to that before the login prompt.

Here is the output of journalctl:
```
Aug 11 20:17:45 matblade systemd[1]: keyboard_login.service: Service is of type D-Bus but no D-Bus service name has been specified. Refusing.
Aug 11 20:21:11 matblade systemd[1]: Started Keyboard random on login.
Aug 11 20:21:11 matblade startup_keyboard[2648]: Unable to init server: Could not connect: Connection refused
Aug 11 20:21:11 matblade startup_keyboard[2648]: Traceback (most recent call last):
Aug 11 20:21:11 matblade startup_keyboard[2648]:   File "/home/mathis/apps/startup_keyboard", line 14, in <module>
Aug 11 20:21:11 matblade startup_keyboard[2648]:     device_manager = DeviceManager()
Aug 11 20:21:11 matblade startup_keyboard[2648]:   File "/usr/lib/python3.6/site-packages/openrazer/client/__init__.py", line 20, in __init__
Aug 11 20:21:11 
```
Here is the keyboard_login.service:
```
[Unit]
Description=Keyboard random on login
Wants=system.slice sysinit.target

[Service]
Type=simple
User=mathis

ExecStart=/home/mathis/apps/startup_keyboard

[Install]
WantedBy=multi-user.target
```
I don't understand why it use in some way the X server, because it doesn't need it for just the keyboard.

If you have any questions, or idea of answer please answer me.
Thank you

z3ntu, 2018-08-11:
First of all, openrazer-daemon is run as a user service so this unit also will have to be a user service (and not a system service).
Second of all, the message literally says that it can't autolaunch a dbus daemon without an X server because it didn't find an existing dbus daemon to connect to, but in general having a dbus daemon without an x server works too.

So try converting it into a user service and try again.

Mathux, 2018-08-11:
Thank you for your quick answer.
Yes, you are right I need to create a user service. 
I tried with 
```
[Unit]
Description=Keyboard random on login
Wants=openrazer-daemon

[Service]
Type=simple

User=mathis
Group=plugdev
ExecStart=/home/mathis/apps/startup_keyboard

[Install]
WantedBy=default.target
```

But I have still an error: (I tried with Group=users and without this line too, same error)

```
● keyboard_login.service - Keyboard random on login
   Loaded: loaded (/home/mathis/.config/systemd/user/keyboard_login.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-08-11 22:58:12 CEST; 14s ago
  Process: 499 ExecStart=/home/mathis/apps/startup_keyboard (code=exited, status=216/GROUP)
 Main PID: 499 (code=exited, status=216/GROUP)
```

z3ntu, 2018-08-11:
Does `journalctl --user-unit keyboard_login.service` say anything useful?

Mathux, 2018-08-17:
I fixed the group issue by removing these lines:
```
User=mathis
Group=plugdev 
```
Now the daemon is working! However because it is a user daemon, my keyboard startup animation begin **after** the login prompt ><. I would rather have that before and stop the animation after opening my X session.

But I don't know if it is really possible to do that because openrazer-daemon is a user daemon.

Do you know a way to do that?

z3ntu, 2018-08-17:
Great, that you got it working!
And no, unfortunately in our current way the daemon is (as you said) starting with the user login. If you really want to do something earlier, you could interact with the sysfs files from the kernel driver (there's a wiki page documenting how to do that). I'm going to close the issue but you can of course still ask questions about whatever you need :)

closed 2018-08-17 by z3ntu

Mathux, 2018-12-24:
Hi @z3ntu,
I finally manage to have my keyboard animation before the opening of the X session ! 
This is maybe a hack but it works and can be useful for other people:
```bash
loginctl enable-linger $USER 
```
This allow the system to load user services before running the session of this user.