openrazer/openrazer #642: Dependency issues with Fedora 26 in Qubes OS

johnschriner, 2018-08-14:
Hi there!  I have a Razor Blade Stealth running Qubes OS.  The dom0 is Fedora 26 and I'm running into dependency issues.  I get no response from a Fedora_26 repo. 

lsusb:
`Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd`

/etc/yum.repos.d/razer.repo
```
[razer]
name = OpenRazer
type = rpm-md
baseurl = http://download.opensuse.org/repositories/hardware:razer/Fedora_27/
enabled = 1
timeout = 100
```
[me@dom0 yum.repos.d]$ sudo qubes-dom0-update openrazer-meta kernel-devel

```
Using sys-firewall as UpdateVM to download updates for Dom0; this may take some time...
Last metadata expiration check: 0:00:00 ago on Tue Aug 14 16:34:19 2018.
Error: 
 Problem: package openrazer-meta-2.3.1-2.1.noarch requires python3-openrazer, but none of the providers can be installed
  - conflicting requests
  - nothing provides python(abi) = 3.6 needed by python3-openrazer-2.3.1-2.1.noarch
```

I didn't find anything about this specific dependency.
Any help would be appreciated and thanks for developing this for years!

z3ntu, 2018-08-14:
The output says that python 3.6 couldn't be found in the repos.
There's no Fedora 26 repo anymore because it is End of Life. You can build the packages from the spec file at https://GitHub.com/OpenRazer/packaging

johnschriner, 2018-08-14:
I'll do that, thanks!

closed 2018-08-15 by z3ntu