openrazer/openrazer #644: Remove code that resets mouse brightness and LEDs on startup

tmitzka, 2018-08-15:
Whenever I adjust brightness and LEDs on my Razer Abyssus v2 mouse, they are reset on startup. I actually opened [an issue](https://github.com/openrazer/openrazer/issues/526) for that problem earlier this year. [These code lines](https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/mouse.py#L821-L824) do that. Thanks again to @z3ntu for pointing me in the right direction.

When I comment these lines out, the problem is solved - for now. But with each update, I have to do it again. I don't really understand the code, so I wondered if there's a chance to just remove that part? Or is there a reason to reset the lights on startup?

closed 2019-01-12 by stale[bot]

sq1000000, 2022-11-29:
Same problem still persists over 4 years later. Would be nice if the devs finally got around to this

z3ntu, 2022-11-29:
@sq1000000 I believe PR #1957 should solve this, if you could test it that'd be helpful.

sq1000000, 2022-11-30:
@z3ntu That branch seems to have solved the issue. The keyboard brightness on my Razer Blade 15 (RZ09-0409) no longer resets to 0 after every reboot or suspend.

But for whatever reason I can't open polychromatic or razercommander anymore. Although, that could just be because of some other processes going on in the background of my arch install. I don't know much about this sort of thing.