openrazer/openrazer #645: Consider a package for Deepin Linux

DanyGee, 2018-08-15:
Trying to install **openrazer-meta_2.3.1-1_all.deb** on **Deepin Linux 15.6** (based on Debian) gives me this error (says I'm missing following packages):

```
openrazer-kernel-modules-dkms
openrazer-daemon
python3-openrazer
openrazer-doc
```

But on this website (meant for Debian)...
https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta
...the **openrazer-meta_2.3.1-1_all.deb** is the **only one available** to download.
How can I install missing packages then? :thinking:

z3ntu, 2018-08-15:
Did you try adding the repo and using apt to install openrazer-meta? The repo contains the packages this package depends on.

DanyGee, 2018-08-15:
Offcourse I tried it.
```
sudo echo 'deb http://download.opensuse.org/repositories/hardware:/razer/Debian_Sid/ /' > /etc/apt/sources.list.d/hardwarerazer.list
bash: /etc/apt/sources.list.d/hardwarerazer.list: Brak dostępu
```
This gives access denied - I don't know why, cause I already have other repos added :thinking: 

Tried the other way:
```
echo 'deb http://download.opensuse.org/repositories/hardware:/razer/Debian_Sid' | sudo tee /etc/apt/sources.list.d/hardware:razer.list
[sudo] hasło użytkownika cyberleszcz:         
deb http://download.opensuse.org/repositories/hardware:/razer/Debian_Sid
```
This adds the repo, but soon as I update...

```
sudo apt-get update
E: Malformed entry 1 in list file /etc/apt/sources.list.d/hardware:razer.list (Suite)
E: Nie udało się odczytać list źródeł. (couldn't read sources list)
```
For the first time I have problem adding a repo. Is there something I'm missing here?
[image]

d0tfile, 2018-08-15:
@DanyGee, you forget ` /`. From software.opensuse.org:
`deb http://download.opensuse.org/repositories/hardware:/razer/Debian_Sid/ /`

DanyGee, 2018-08-15:
> @DanyGee, you forget /. From software.opensuse.org:
deb http://download.opensuse.org/repositories/hardware:/razer/Debian_Sid/ /

That was the cause of the problem! Case closed.
THX :champagne:

closed 2018-08-15 by DanyGee