openrazer/openrazer #648: Orbweaver Chroma no key macro change options

macasio14, 2018-08-20:
I'm suspecting this is a limitation in openrazer, but the only option the polychromatic gives me for the orbweaver is to change the brightness and lighting effects, and very limited lighting effects at that.

Can someone confirm if this is openrazer or polychromatic problem?

lah7, 2018-08-21:
It's a limitation in OpenRazer for now. At the moment, once the daemon initialises the keypad, it behaves like a Chroma keyboard due to the history of how it was implemented.

macasio14, 2018-08-21:
Anything I can do to help getting it working? My knowledge in code is limited, I could be effective in testing or gathering wireshark data if that would be useful?

macasio14, 2018-08-25:
I just tried a usb mointoring program, https://www.eltima.com/products/usb-port-monitor/ demo version. Regardless of even setting to monitor all of my connected usb peripherals, nothing was coming over usb that the program was recognizing. Going to see if I can pester Razer directly in some way shape or form for more information. **Since Openrazer does appear to have the capabilities to remap some keyboard keys (please correct me if I'm wrong)** so this should be well in the range of doing the same thing.

z3ntu, 2018-08-25:
OpenRazer can't remap keys for now.

macasio14, 2018-08-25:
Ok, I didn't understand the full functionality of OpenRazer then. Thank you @z3ntu. Should I keep this issue open, or close it?

z3ntu, 2018-08-25:
I think there are enough issues open about key rebinding, so I'll close this :)

closed 2018-08-25 by z3ntu