openrazer/openrazer #650: Device request: Naga Epic MMO

jon2kx, 2018-08-21:
Naga Epic would be a nice to have. I have had one for a long time.

Strowser, 2018-08-27:
Agree! A classic, that I also have, would be a nice addition.

z3ntu, 2018-09-14:
* Please post the output of
`lsusb -d 1532: -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

MrDick47, 2019-02-09:
Razer Naga Epic Chroma
Model Number: RC30-012301
Product Number: RZ01-01230100

I'm using Debian 9, I also have another laptop running Windows 10 with Synapse and such so I will get those screenshots and will try to capture some packets (unless there is a problem with capturing packets on another machine, I could setup a VM in that case).

These were taken in WIRED mode! This mouse also has a wireless mode with its charging dock, if needed I can try to find mine and run the commands/capture packets using that later.

Here is the output of `sudo lsusb -d 1532: -v`
[razer-naga-epic-lsusb.txt](https://github.com/openrazer/openrazer/files/2848035/razer-naga-epic-lsusb.txt)

```
Bus 001 Device 008: ID 1532:003e Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x003e 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Epic Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
And here is the output of `sudo ls -lah /dev/input/by-id`
[razer-naga-epic-ls-dev-input.txt](https://github.com/openrazer/openrazer/files/2848034/razer-naga-epic-ls-dev-input.txt)
```
total 0
drwxr-xr-x 2 root root 160 Feb  9 10:52 .
drwxr-xr-x 4 root root 700 Feb  9 10:52 ..
lrwxrwxrwx 1 root root  10 Feb  9 10:25 usb-DETNR019IA30N4_HP_HD_Camera_200901010001-event-if00 -> ../event19
lrwxrwxrwx 1 root root   9 Feb  9 10:47 usb-Razer_Razer_Naga_Epic_Chroma-event-if02 -> ../event4
lrwxrwxrwx 1 root root   9 Feb  9 10:47 usb-Razer_Razer_Naga_Epic_Chroma-event-mouse -> ../event1
lrwxrwxrwx 1 root root   9 Feb  9 10:47 usb-Razer_Razer_Naga_Epic_Chroma-if01-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Feb  9 10:47 usb-Razer_Razer_Naga_Epic_Chroma-if02-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Feb  9 10:47 usb-Razer_Razer_Naga_Epic_Chroma-mouse -> ../mouse0
```

Please let me what I can do to help in any way, as I am a fellow programmer/engineer! :)

z3ntu, 2019-03-20:
Duplicate of #299

closed 2019-03-20 by z3ntu

ChaoticAce, 2019-08-24:
This is still an issue. The device shows up in the main package, and I run into a BS when I try to build from the feature branch on manjaro when installing from the AUR