openrazer/openrazer #654: openrazer wont open..

pranayama26, 2018-08-23:
`● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-08-23 05:35:11 CDT; 24s ago
     Docs: man:openrazer-daemon(8)
  Process: 1024 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 1024 (code=exited, status=1/FAILURE)

Aug 23 05:35:11 anarchy systemd[592]: Starting Daemon to manage razer devices in userspace...
Aug 23 05:35:11 anarchy openrazer-daemon[1024]: Traceback (most recent call last):
Aug 23 05:35:11 anarchy openrazer-daemon[1024]:   File "/usr/bin/openrazer-daemon", line 11, in <module>
Aug 23 05:35:11 anarchy openrazer-daemon[1024]:     from openrazer_daemon.daemon import RazerDaemon, __version__
Aug 23 05:35:11 anarchy openrazer-daemon[1024]: ModuleNotFoundError: No module named 'openrazer_daemon'
Aug 23 05:35:11 anarchy systemd[592]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Aug 23 05:35:11 anarchy systemd[592]: openrazer-daemon.service: Failed with result 'exit-code'.
Aug 23 05:35:11 anarchy systemd[592]: Failed to start Daemon to manage razer devices in userspace.

`

z3ntu, 2018-08-23:
If this is Arch Linux, you'll have to rebuild the python-notify2 and openrazer-meta packages because of the recent upgrade to Python 3.7.

pranayama26, 2018-08-23:
How do i do that, i tried and reinstalling it but that did not work.

pranayama26, 2018-08-23:
> makepkg -srifCc in the openrazer-meta (and python-notify2) directory.

So i need to down the the git clone then

kanarok, 2018-08-30:
If you are on arch you can install from aur/python-openrazer aur/openrazer-daemon aur/openrazer-driver-dkms e.g. with yaourt. This works for me.

z3ntu, 2018-09-25:
Did you manage to fix it?

z3ntu, 2018-12-03:
Closing as there hasn't been a response.

closed 2018-12-03 by z3ntu