openrazer/openrazer #655: Startup Fails "Background Service Could not be started"

jappyjan, 2018-08-25:
I installed everything as described by "Readme.md", rebooted and afterwards tryed to run the application.

For 10 Seconds a loading spinner comes up and then it says "Dienst läuft nicht" (German Version) which means the Service is not running.

Pressing the "Restart Background Service" button didn't help.

z3ntu, 2018-08-25:
What's the output of `systemctl --user status openrazer-daemon.service` ?

jappyjan, 2018-08-25:
The output is this:

`jappy@Jappy-Switch:~$ systemctl --user status openrazer-daemon.service
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-08-25 15:36:13 CEST; 7min ago
     Docs: man:openrazer-daemon(8)
  Process: 2392 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 2392 (code=exited, status=1/FAILURE)

Aug 25 15:36:13 Jappy-Switch openrazer-daemon[2392]: Traceback (most recent call last):
Aug 25 15:36:13 Jappy-Switch openrazer-daemon[2392]:   File "/usr/bin/openrazer-daemon", line 177, in <module>
Aug 25 15:36:13 Jappy-Switch openrazer-daemon[2392]:     run()
Aug 25 15:36:13 Jappy-Switch openrazer-daemon[2392]:   File "/usr/bin/openrazer-daemon", line 172, in run
Aug 25 15:36:13 Jappy-Switch openrazer-daemon[2392]:     logger=logger)
Aug 25 15:36:13 Jappy-Switch openrazer-daemon[2392]: TypeError: __init__() got an unexpected keyword argument 'foreground'
Aug 25 15:36:13 Jappy-Switch systemd[1538]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Aug 25 15:36:13 Jappy-Switch systemd[1538]: Failed to start Daemon to manage razer devices in userspace.
Aug 25 15:36:13 Jappy-Switch systemd[1538]: openrazer-daemon.service: Unit entered failed state.
Aug 25 15:36:13 Jappy-Switch systemd[1538]: openrazer-daemon.service: Failed with result 'exit-code'.
`

z3ntu, 2018-08-25:
Didn't you add the ppa mentioned on the website? If you did, do an `sudo apt update && sudo apt upgrade` and then try again.

jappyjan, 2018-08-25:
i tryed, it fails (maybe because im using debian, im very new to the linux ecosystem) so i installed everything by .deb packages

z3ntu, 2018-08-25:
Please follow https://openrazer.github.io/#debian . This will tell you everything you have to do.

Also thumb rule: Don't install raw .deb packages unless there's no better way (e.g. adding a repo) :)

jappyjan, 2018-08-25:
still the same

z3ntu, 2018-08-25:
Did you do an `sudo apt update && sudo apt upgrade` after adding the repo?

closed 2018-09-25 by z3ntu