openrazer/openrazer #656: daemon will not start

thranmorrow, 2018-08-27:
I am running ubuntu 18.04 and installed using the instructions found here https://openrazer.github.io/
when I try to start the driver daemon it throws an error.

Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 177, in <module>
    run()
  File "/usr/bin/openrazer-daemon", line 172, in run
    logger=logger)
TypeError: __init__() got an unexpected keyword argument 'foreground'

what am I doing wrong?

z3ntu, 2018-08-27:
Please run `sudo apt update && sudo apt upgrade` and then try again.

thranmorrow, 2018-08-27:
sudo apt update && sudo apt upgrade executed
by try again do you mean install it again or run openrazer-daemon -Fv again?
I did the later and go the same error

z3ntu, 2018-08-27:
Did you properly add the openrazer ppa?

thranmorrow, 2018-08-27:
I added it using sudo add-apt-repository ppa:openrazer/daily

z3ntu, 2018-08-27:
Okay our daily PPA is missing python3-daemonize currently. But you should be fine installing the version for 17.10 (artful) which you can get at https://launchpad.net/~openrazer/+archive/ubuntu/daily/+files/python3-daemonize_2.4.7-artful_all.deb
(install it with `dpkg -i filename`)

thranmorrow, 2018-08-27:
I already had python3-daemonize installed
{code}
sudo apt list --installed | grep python3-daemonize

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python3-daemonize/artful,artful,now 2.3.1-1 all [installed]
{code}

z3ntu, 2018-08-27:
Okay nvm, bionic (18.04) has python3-daemonize version 2.4.7-2 in their repos. Are you sure, you're on 18.04?

thranmorrow, 2018-08-27:
certian

cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04 LTS

z3ntu, 2018-08-27:
Your apt output from before (`python3-daemonize/artful,artful,now 2.3.1-1 all [installed]`) says `artful` though which is the codename for Ubuntu 17.10 so not all packages were upgraded or something.

thranmorrow, 2018-08-27:
I uninstalled the python3-daemonize earlier today and reinstalled. artful is what I got. and looking at the repos  2.3.1-1 is the latest stable release

z3ntu, 2018-08-27:
https://packages.ubuntu.com/bionic/python3-daemonize