openrazer/openrazer #663: Settings not persistent, lost after reboot

APoniatowski, 2018-09-01:
Hi there,

there seems to be a problem with keeping the settings persistent. LED settings are lost and the Turn off on screensaver setting is also lost and defaults to being enabled, which is quite annoying in the dark (black keyboard in the dark). Polling rate is defaulted to 500Hz as well.

Is there a way to make this permanent? like a quick workaround?

I am using Manjaro KDE:
Screenfetch

██████████████████ ████████ OS: Manjaro 17.1.12 Hakoila
██████████████████ ████████ Kernel: x86_64 Linux 4.14.66-1-MANJARO
██████████████████ ████████ Uptime: 3h 29m
████████ ████████ Packages: 1121
████████ ████████ ████████ Shell: bash 4.4.23
████████ ████████ ████████ Resolution: 4480x1440
████████ ████████ ████████ DE: KDE 5.49.0 / Plasma 5.13.4
████████ ████████ ████████ WM: KWin
████████ ████████ ████████ WM Theme: Breeze Dark
████████ ████████ ████████ GTK Theme: Breath [GTK2/3]
████████ ████████ ████████ Icon Theme: maia
████████ ████████ ████████ Font: Noto Sans Regular
████████ ████████ ████████ CPU: Intel Core i7-4770K @ 8x 3.5GHz [37.0°C]
GPU: GeForce GTX 1070
RAM: 2816MiB / 15982MiB

z3ntu, 2018-09-01:
As I wrote in the RazerGenie repo it's not easily implemented with the current architecture :)

lah7, 2018-09-01:
A workaround could be to create a start-up Python script that interacts with the OpenRazer Python library directly, like this:

```
#!/bin/bash
import openrazer.client
a = openrazer.client.DeviceManager()
a.turn_off_on_screensaver = False
for device in a.devices:
    device.brightness = 100
    device.fx.static(0, 255, 0)
    if device.type == "mouse":
        device.poll_rate = 500
```

:point_up: May vary depending on your device, you can inspect and test by typing the commands in the `python3` console and pressing tab to see the combinations.

APoniatowski, 2018-09-01:
Thanks guys :)  I will implement a python script to keep the LED colors persistent :)

About the little script you wrote lah7... don't you mean "#!/bin/python3" ?  ;)

lah7, 2018-09-01:
Whoops, good spot! I'll amend the comment.

APoniatowski, 2018-09-12:
I've been playing around the effects for my KB, that's fine. for fx and type of fx, how to I narrow it down for my mouse?  the docs makes sense for for kb, but not a mouse though.

Got a naga hex v2 mouse

lah7, 2018-11-02:
RazerGenie interacts with the daemon directly, so it might not have installed the python libraries which will be needed for that script to work. On Arch, it's packaged under `python-openrazer`, make sure this is installed.

@z3ntu packages for Arch, he'll be able to confirm the versioning for `-git` .

APoniatowski, 2018-11-02:
funny enough, I have that installed... even tried openrazer lib via pip, still no-go :(

APoniatowski, 2018-11-08:
updated openrazer-meta now.

I think I found my problem... you were right @lah7  
$ which python
/home/{username}/anaconda3/bin/python
$ which python3
/home/{username}/anaconda3/bin/python3

z3ntu, 2018-11-08:
It should work though when you use it with the `#!/usr/bin/python3` shebang. How do you invoke the script?

APoniatowski, 2018-11-09:
via #!/usr/bin/python3... which is strange.

APoniatowski, 2018-11-16:
okey dokey... selectively blind me... I've been trying to run it with "python my-script.py" and not give it execution permissions and then running it normally, as it was intended (fun times with dual booting and forgetting what you did in linux lol)

z3ntu, 2018-11-17:
So it's working now for you?

APoniatowski, 2018-11-20:
yeah, my script is working now... I was being daft when running my script. OS-hopping sucks, it makes me forget what I was doing in OS I log back into

closed 2018-11-20 by z3ntu

komidawi, 2019-11-04:
> A workaround could be to create a start-up Python script that interacts with the OpenRazer Python library directly, like this:
> 
> ```
> #!/usr/bin/python3
> import openrazer.client
> a = openrazer.client.DeviceManager()
> a.turn_off_on_screensaver = False
> for device in a.devices:
>     device.brightness = 100
>     device.fx.static(0, 255, 0)
>     if device.type == "mouse":
>         device.poll_rate = 500
> ```

I managed to set DPI, but how to make my logo lightning in breath mode?
_(Razer Deathadder Essential 6400dpi)_

lah7, 2019-11-04:
@komidawi Probably:

    device.fx.breath_random()
    device.fx.breath_single(0, 255, 0)
    device.fx.breath_dual(0, 255, 0, 255, 0 0)

Or `fx.misc.logo` instead of just `fx`. I don't have this mouse to confirm the exact function. You can poke about using the `python3` interpreter.

komidawi, 2019-11-04:
@lah7 thank you very much, `fx.misc.logo.breath_single(0, 255, 0)` worked like a charm :smiley_cat:

lah7, 2022-11-07:
@jfaz1 Frontends have no responsibility over persistence. OpenRazer (>=3.0.0) has had a software persistence feature for a while now that's supposed to apply the settings at logon instead of a relying on a script.

Check this file:  `~/.config/openrazer/razer.conf`
```ini
[Startup]
restore_persistence = True
```

Next to that is `persistence.conf` which stores the settings:
```ini
[PMXXXXXXXXXX123]
dpi_x = 1800
dpi_y = 1800
poll_rate = 500
```

If it doesn't work when it's on, try turning it off (`restore_persistence = False` - when off, no command is sent. It would be entirely up to the mouse to retain the poll rate). If there's still problems, please raise an issue for your particular mouse - it could be a bug.