openrazer/openrazer #666: Orbweaver Chroma custom effects

CalcProgrammer1, 2018-09-03:
I modified the driver by adding the custom effect and custom frame entries to the Orbweaver Chroma probe and disconnect functions. The mapping is not 1 to 1 so I did some tests to map it out.  I would say leave the mapping up to the application/daemon but that's up to the project maintainers I guess.

Attached is my mapping.  The number in the box is the Orbweaver key in that row/col.
[image]

sevu, 2018-09-30:
Hi @CalcProgrammer1, another Orbweaver Chroma user here.

What is this about? Are in the middle the keys on the keyboard, and the one gap is on German keyboard Layout for the  "<" key? What are the numbers at the borders representing?
What are the "custom effect and custom frame entries" ?

Maybe you would like to make a Pull Request?

chrisdonlan87, 2018-12-03:
Macros for the orb weaver are not yet supported, right?

closed 2019-02-28 by stale[bot]

hedmo, 2019-12-09:
how about wave for the orbweaver?