openrazer/openrazer #667: Razer Kraken 7.1 V2 not available anymore in the Controller

Konpon96, 2018-09-05:
Just noticed that my Razer Kraken 7.1 V2 headphones can't be managed through Polychromatic. Their LEDs also don't turn off when the screen is locked, although my Razer's keyboard and mouse do so (BlackWidow Chroma V2, DeathAdder Elite).

My output from lsusb with Razer's vendor ID grep is as follows:

`Bus 005 Device 002: ID 1532:0221 Razer USA, Ltd `
`Bus 001 Device 005: ID 1532:0510 Razer USA, Ltd `
`Bus 001 Device 003: ID 1532:005c Razer USA, Ltd `

I'm using Arch Linux (Polychromatic installed from AUR) with GNOME and GDM, here is my uname output:

`Linux 4.18.5-arch1-1-ARCH #1 SMP PREEMPT Fri Aug 24 12:48:58 UTC 2018 x86_64 GNU/Linux`

z3ntu, 2018-09-05:
`ls -al /sys/bus/hid/drivers/razerkraken/0*/` ?

Konpon96, 2018-09-05:
Here is the output:

```
drwxr-xr-x 5 root root    0 09-05 13:36 .
drwxr-xr-x 6 root root    0 09-05 13:36 ..
-r--r--r-- 1 root root 4096 09-05 17:37 country
-rw-rw---- 1 root root 4096 09-05 17:37 device_mode
-r--r----- 1 root root 4096 09-05 17:37 device_serial
-r--r----- 1 root root 4096 09-05 17:37 device_type
lrwxrwxrwx 1 root root    0 09-05 13:36 driver -> ../../../../../../../bus/hid/drivers/razerkraken
-r--r----- 1 root root 4096 09-05 17:37 firmware_version
drwxr-xr-x 3 root root    0 09-05 13:36 hidraw
drwxr-xr-x 3 root root    0 09-05 13:36 input
-r--r----- 1 root root 4096 09-05 17:37 matrix_current_effect
-rw-rw---- 1 root root 4096 09-05 17:37 matrix_effect_breath
-rw-rw---- 1 root root 4096 09-05 17:37 matrix_effect_custom
--w--w---- 1 root root 4096 09-05 17:37 matrix_effect_none
--w--w---- 1 root root 4096 09-05 17:37 matrix_effect_spectrum
-rw-rw---- 1 root root 4096 09-05 17:37 matrix_effect_static
-r--r--r-- 1 root root 4096 09-05 17:37 modalias
drwxr-xr-x 2 root root    0 09-05 17:37 power
-r--r--r-- 1 root root 4096 09-05 17:37 report_descriptor
lrwxrwxrwx 1 root root    0 09-05 13:36 subsystem -> ../../../../../../../bus/hid
-rw-rw---- 1 root root 4096 09-05 17:37 test
-rw-r--r-- 1 root root 4096 09-05 17:37 uevent
-r--r----- 1 root root 4096 09-05 17:37 version
```

z3ntu, 2018-09-05:
Please upload `~/.local/share/openrazer/logs/razer.log` to a paste service or attach it as a text file here.

Konpon96, 2018-09-07:
Here is my log file:
[razer.log](https://github.com/openrazer/openrazer/files/2359180/razer.log)

What is very interesting is that the headphones worked once, one can find the lines:
```
2018-09-07 03:11:38 | razer                          | INFO     | Found device.1: 0003:1532:0510.0008
2018-09-07 03:11:38 | razer.device1                  | INFO     | Initialising device.1 RazerKrakenV2
```

After this my headphones were available in the Controller, their LEDs were also turning off when the screensaver was on. After rebooting they are not available anymore and the following line appeared:
```
2018-09-07 03:20:18 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.3/0003:1532:0510.0008/device_type, file is not owned by plugdev
```

The file is owned by root, just as it can be seen on the previous output I posted. My user is member of a plugdev group, but this looks like a problem with the pseudofile itself.

For the comparison purposes, here is my output of `ls -al /sys/bus/hid/drivers/razerkbd/0*/` (I trimmed it to one catalogue but it can be seen that all the files have group set to plugdev):

```
'/sys/bus/hid/drivers/razerkbd/0003:1532:0221.0003/':
total 0
drwxr-xr-x 5 root plugdev    0 09-07 03:20 .
drwxr-xr-x 5 root root       0 09-07 03:19 ..
-r--r--r-- 1 root plugdev 4096 09-07 03:20 country
-rw-rw---- 1 root plugdev 4096 09-07 03:20 device_mode
-r--r----- 1 root plugdev 4096 09-07 03:20 device_serial
-r--r----- 1 root plugdev 4096 09-07 03:20 device_type
lrwxrwxrwx 1 root plugdev    0 09-07 03:20 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 09-07 03:20 firmware_version
-rw-rw---- 1 root plugdev 4096 09-07 03:20 game_led_state
drwxr-xr-x 3 root plugdev    0 09-07 03:20 hidraw
drwxr-xr-x 3 root plugdev    0 09-07 03:20 input
-r--r----- 1 root plugdev 4096 09-07 03:20 kbd_layout
-rw-rw---- 1 root plugdev 4096 09-07 03:20 macro_led_effect
-rw-rw---- 1 root plugdev 4096 09-07 03:20 macro_led_state
-rw-rw---- 1 root plugdev 4096 09-07 03:20 matrix_brightness
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_custom_frame
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_effect_breath
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_effect_custom
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_effect_none
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_effect_reactive
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_effect_starlight
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_effect_static
--w--w---- 1 root plugdev 4096 09-07 03:20 matrix_effect_wave
-r--r--r-- 1 root plugdev 4096 09-07 03:20 modalias
drwxr-xr-x 2 root plugdev    0 09-07 03:20 power
-r--r--r-- 1 root plugdev 4096 09-07 03:20 report_descriptor
lrwxrwxrwx 1 root plugdev    0 09-07 03:20 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 09-07 03:20 test
-rw-r--r-- 1 root plugdev 4096 09-07 03:20 uevent
-r--r----- 1 root plugdev 4096 09-07 03:20 version
```

Konpon96, 2018-09-07:
I tried reinstalling all the components, from openrazer to polychromatic and it didn't help - neither rebooting did. Unplugging and plugging the headphones worked though.

Here is my journalctl output:

```
wrz 07 14:53:06 kppc razer_mount[625]: Driver razerkbd
wrz 07 14:53:06 kppc razer_mount[624]: Driver razerkbd
wrz 07 14:53:06 kppc razer_mount[623]: Driver razerkbd
wrz 07 14:53:06 kppc razer_mount[626]: Device_ID 0003:1532:0221.0003
wrz 07 14:53:06 kppc razer_mount[627]: Device_ID 0003:1532:0221.0002
wrz 07 14:53:06 kppc razer_mount[628]: Device_ID 0003:1532:0221.0001
wrz 07 14:53:06 kppc razer_mount[630]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0003/
wrz 07 14:53:06 kppc razer_mount[631]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/
wrz 07 14:53:06 kppc razer_mount[632]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0001/
wrz 07 14:53:06 kppc razer_mount[639]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0003/
wrz 07 14:53:06 kppc razer_mount[638]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0001/
wrz 07 14:53:06 kppc razer_mount[640]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/
wrz 07 14:53:07 kppc razer_mount[888]: Driver razerkraken
wrz 07 14:53:07 kppc razer_mount[889]: Device_ID 0003:1532:0510.0008
wrz 07 14:53:07 kppc razer_mount[895]: Driver razermouse
wrz 07 14:53:07 kppc razer_mount[896]: Driver razermouse
wrz 07 14:53:07 kppc razer_mount[897]: Driver razermouse
wrz 07 14:53:07 kppc razer_mount[898]: Device_ID 0003:1532:005C.0004
wrz 07 14:53:07 kppc razer_mount[899]: Device_ID 0003:1532:005C.0006
wrz 07 14:53:07 kppc razer_mount[900]: Device_ID 0003:1532:005C.0005
wrz 07 14:53:07 kppc razer_mount[901]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0004/
wrz 07 14:53:07 kppc razer_mount[902]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0006/
wrz 07 14:53:07 kppc razer_mount[903]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0005/
wrz 07 14:53:07 kppc razer_mount[909]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0004/
wrz 07 14:53:07 kppc razer_mount[910]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0006/
wrz 07 14:53:07 kppc razer_mount[911]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0005/
wrz 07 18:37:22 kppc razer_mount[8991]: Driver razerkraken
wrz 07 18:37:22 kppc razer_mount[8993]: Device_ID 0003:1532:0510.000A
wrz 07 18:37:22 kppc razer_mount[8994]: Changing group /sys/bus/hid/drivers/razerkraken/0003:1532:0510.000A/
wrz 07 18:37:22 kppc razer_mount[8996]: Changed group /sys/bus/hid/drivers/razerkraken/0003:1532:0510.000A/
```

It's clearly written that after booting keyboard and mouse get detected and groups of their respective catalogues in /sys get chown-ed. Headphones (razerkraken) get detected too. However, their /sys catalogues don't get chown-ed. I worked for a few hours, then replugged headphones and chown has been invoked successfully - Polychromatic Controller works as well.

Here is the output of /sys catalogue of the headphones after replugging:

```
drwxr-xr-x 5 root plugdev    0 09-07 18:37 .
drwxr-xr-x 6 root root       0 09-07 18:37 ..
-r--r--r-- 1 root plugdev 4096 09-07 18:37 country
-rw-rw---- 1 root plugdev 4096 09-07 18:37 device_mode
-r--r----- 1 root plugdev 4096 09-07 18:37 device_serial
-r--r----- 1 root plugdev 4096 09-07 18:37 device_type
lrwxrwxrwx 1 root plugdev    0 09-07 18:37 driver -> ../../../../../../../bus/hid/drivers/razerkraken
-r--r----- 1 root plugdev 4096 09-07 18:37 firmware_version
drwxr-xr-x 3 root plugdev    0 09-07 18:37 hidraw
drwxr-xr-x 3 root plugdev    0 09-07 18:37 input
-r--r----- 1 root plugdev 4096 09-07 18:37 matrix_current_effect
-rw-rw---- 1 root plugdev 4096 09-07 18:37 matrix_effect_breath
-rw-rw---- 1 root plugdev 4096 09-07 18:37 matrix_effect_custom
--w--w---- 1 root plugdev 4096 09-07 18:40 matrix_effect_none
--w--w---- 1 root plugdev 4096 09-07 18:40 matrix_effect_spectrum
-rw-rw---- 1 root plugdev 4096 09-07 18:37 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 09-07 18:37 modalias
drwxr-xr-x 2 root plugdev    0 09-07 18:37 power
-r--r--r-- 1 root plugdev 4096 09-07 18:37 report_descriptor
lrwxrwxrwx 1 root plugdev    0 09-07 18:37 subsystem -> ../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 09-07 18:37 test
-rw-r--r-- 1 root plugdev 4096 09-07 18:37 uevent
-r--r----- 1 root plugdev 4096 09-07 18:37 version
```

Looks like the problem occurs only when headphones are plugged when the system is booting.

closed 2019-02-05 by stale[bot]