openrazer/openrazer #669: Unresolved dependencies during installation

ServantGrunt, 2018-09-07:
Good day, I'm on Manjaro (Arch Linux based distro) using KDE Plasma as graphic environment and when trying to install the openrazer software via `makepkg -si` (after `git clone`-ing the repository from the AUR upstream) I get some unresolved dependencies:

> :: unable to satisfy dependency 'python-notify2' required by openrazer-daemon
:: unable to satisfy dependency 'openrazer-daemon' required by python-openrazer
:: unable to satisfy dependency 'python-notify2' required by openrazer-daemon
:: unable to satisfy dependency 'python-notify2' required by openrazer-daemon
:: unable to satisfy dependency 'openrazer-daemon' required by openrazer-meta
:: unable to satisfy dependency 'python-notify2' required by openrazer-daemon
:: unable to satisfy dependency 'openrazer-daemon' required by python-openrazer
:: unable to satisfy dependency 'python-openrazer' required by openrazer-meta

so I can't compile python-openrazer, openrazer-daemon and openrazer-meta. Is there anything I can do to fix this? Thanks in advance.

z3ntu, 2018-09-07:
Install python-notify2 from the AUR.

closed 2018-09-07 by z3ntu

dbuehler85, 2020-10-08:
That package is now python2-notify2.
The original package is no longer available.

lah7, 2020-10-08:
It appears [`python-notify2`](https://www.archlinux.org/packages/extra/any/python-notify2/) moved to the official Arch repository ([2020-10-02](https://github.com/archlinux/svntogit-packages/commits/packages/python-notify2/trunk)), but Manjaro's stable repositories (2020-10-01) won't include this until the next stable update.

The workaround would be to switch to Manjaro's testing branch, or download the [`python-notify2` package from the mirror.](https://www.archlinux.org/packages/extra/any/python-notify2/download/)