openrazer/openrazer #670: Device not Recognised

Ultracard, 2018-09-07:
Hello, I'm new to Linux and I'm not very experienced with it.
I installed the Openrazer driver, and the daemon seems to work. But i downloaded Polychromatic and it shows me an error message, that my device was not recognised. Even though i have a Blackwidow Chroma V2 which is supported. I also have all the dependencies installed.
Can somebody help me? Because i don't know what the issue is. I attached my Razer log if that helps to find the issue.
[razer.log](https://github.com/openrazer/openrazer/files/2359985/razer.log)
Thank You!

z3ntu, 2018-09-07:
Please read the first line (labeled `CRITICAL`) and then run `sudo gpasswd -a $USER plugdev` to add your user to the plugdev group.

Ultracard, 2018-09-07:
Yes i already added myself to the plugdev group.

Ultracard, 2018-09-07:
When i run sudo dkms install openrazer-driver/2.3.1
i get an Error, that my kernel headers for kernel 4.15.0-kali2-amd cannot be foun.

z3ntu, 2018-09-07:
Then install the kernel headers for your kernel.

Ultracard, 2018-09-07:
But that might be because yesterday i already tried fixing the problem and on another issue somebody said that i should run 
**sudo apt purge openrazer-meta polychromatic linux-headers-generic**
did that create a problem on my Computer?

Ultracard, 2018-09-07:
Yes i tried, but when i run
**sudo apt-get install linux-headers-generic**
i get this message in the Terminal:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package linux-headers-generic is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'linux-headers-generic' has no installation candidate

z3ntu, 2018-09-07:
On Kali Linux the package name is probably different.

Ultracard, 2018-09-07:
I found these commands on the internet, do you think I should try them?
**sudo apt-get dist-upgrade -y 
sudo apt-get install linux-headers-$(uname -r)
sudo apt-get upgrade linux-headers-$(uname -r)**

z3ntu, 2018-09-07:
`sudo apt-get install linux-headers-$(uname -r)` should do the job (but run the dist-upgrade command before)

Ultracard, 2018-09-07:
Ok,it's currently upgrading, it will take a few minutes.

Ultracard, 2018-09-07:
when i run
**sudo apt-get install linux-headers-$(uname -r)**
i get:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.15.0-kali2-amd64
E: Couldn't find any package by glob 'linux-headers-4.15.0-kali2-amd64'
E: Couldn't find any package by regex 'linux-headers-4.15.0-kali2-amd64'

Ultracard, 2018-09-07:
I just reebootet  my pc, after entering my password I only get a blank screen, I guess I will have to install linux again, but thank you so much, I would have never thought that I would get help so quickly ☺️.

z3ntu, 2018-09-07:
Sure.

closed 2018-09-07 by z3ntu