openrazer/openrazer #671: Installation doesn't work on Fedora 28

wmouchere, 2018-09-07:
Hello, I've tried installing openrazer on Fedora 28 but it doesn't seem to work.
Peripherals are detected but not by the daemon.

```
dmesg | grep -i razer
[    1.257537] usb 1-8: Product: Razer Kraken 7.1 Chroma
[    1.257538] usb 1-8: Manufacturer: Razer
[    1.340635] input: Razer Razer Kraken 7.1 Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.3/0003:1532:0504.0001/input/input6
[    1.392232] hid-generic 0003:1532:0504.0001: input,hiddev96,hidraw0: USB HID v1.11 Device [Razer Razer Kraken 7.1 Chroma] on usb-0000:00:14.0-8/input3
[    1.642240] usb 1-9: Product: Razer BlackWidow Chroma
[    1.642241] usb 1-9: Manufacturer: Razer
[    1.643634] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:1532:0203.0002/input/input7
[    1.695237] hid-generic 0003:1532:0203.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma] on usb-0000:00:14.0-9/input0
[    1.695897] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.1/0003:1532:0203.0003/input/input8
[    1.747664] hid-generic 0003:1532:0203.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma] on usb-0000:00:14.0-9/input1
[    1.748177] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.2/0003:1532:0203.0004/input/input9
[    1.748543] hid-generic 0003:1532:0203.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer BlackWidow Chroma] on usb-0000:00:14.0-9/input2
[    1.990233] usb 1-10: Product: Razer DeathAdder 2013
[    1.990233] usb 1-10: Manufacturer: Razer
[    1.992241] input: Razer Razer DeathAdder 2013 as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:1532:0037.0005/input/input10
[    1.992387] hid-generic 0003:1532:0037.0005: input,hidraw4: USB HID v1.11 Mouse [Razer Razer DeathAdder 2013] on usb-0000:00:14.0-10/input0
[    1.993887] input: Razer Razer DeathAdder 2013 as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:1532:0037.0006/input/input11
[    2.045215] hid-generic 0003:1532:0037.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer DeathAdder 2013] on usb-0000:00:14.0-10/input1
[    2.045796] input: Razer Razer DeathAdder 2013 as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:1532:0037.0007/input/input12
[    2.097216] hid-generic 0003:1532:0037.0007: input,hidraw6: USB HID v1.11 Keyboard [Razer Razer DeathAdder 2013] on usb-0000:00:14.0-10/input2

```
Daemon log :

```
2018-09-07 09:53:35 | razer                          | INFO     | Initialising Daemon (v2.3.1). Pid: 4294
2018-09-07 09:53:35 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerCynosaChroma ----------------------- (1532:022A)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2018-09-07 09:53:35 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
[... 40 lines cut ...]
Sep 07 09:41:35 localhost.localdomain razer_mount[650]: Driver razerkbd
[... cut ...]
Sep 07 09:41:35 localhost.localdomain systemd-udevd[608]: Process 'razer_mount razerkbd 0003:1532:0203.0002' failed with exit code 1.
Sep 07 09:41:35 localhost.localdomain razer_mount[667]: Modprobing razerkbd
Sep 07 09:41:35 localhost.localdomain razer_mount[662]: Modprobing razerkraken
```

I tried (re)installing the driver : 

```
sudo dmks install openrazer-driver/2.3.1
Module openrazer-driver/2.3.1 already installed on kernel 4.17.19-200.fc28.x86_64/x86_64
```

I know that my Deathadder 2013 is not supported, the rest should be ok though.
Thanks for you help !

z3ntu, 2018-09-07:
Please run
`sudo dkms remove openrazer-driver/2.3.1 --all`
and
`sudo dkms install openrazer-driver/2.3.1`

wmouchere, 2018-09-07:
I ran remove / install without success (went fine but results were the same)

```
sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Operation not permitted
```

z3ntu, 2018-09-07:
Anything in the dmesg related to the modprobe call?

wmouchere, 2018-09-07:
Secure boot was enabled, that was the problem.
Any way to sign the module so that I can enable it back ?

z3ntu, 2018-09-07:
Interesting that secure boot produced that message. Yes, you can sign the dkms modules, but I can't tell you how. Ask your favorite search engine for that.

closed 2018-09-07 by z3ntu