openrazer/openrazer #674: Support for the deathadder expert

ripper81, 2018-09-11:
I have an razer deathadder gamingmouse. Unfortunately there seems no support for it by the openrazer tools right now. Is there a chance that this fact could change ?

z3ntu, 2018-09-14:
* Please post the output of
`lsusb -d 1532: -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

gabrielcaldeira757, 2018-11-01:
Hi, my name is gabriel and I have the same mouse.
I really wanted the support for this mouse and I'm willing to help, I'm a python programmer, if this helps, what is nescessario for this support to be commited ??

bobrowich, 2018-11-24:
Can you add support for deathadder expert?
Asked output is:

- 8:Bus 003 Device 002: ID 1532:0037 Razer USA, Ltd

- usb-Razer_Razer_DeathAdder_2013-event-mouse -> ../event12
- usb-Razer_Razer_DeathAdder_2013-if01-event-kbd -> ../event13
- usb-Razer_Razer_DeathAdder_2013-if02-event-kbd -> ../event14
- usb-Razer_Razer_DeathAdder_2013-mouse -> ../mouse1

z3ntu, 2018-11-25:
@bobrowich Your mouse is the "DeathAdder 2013", see #195 for that.

@gabrielcaldeira757 @ripper81 Please get me the lsusb output as requested in https://github.com/openrazer/openrazer/issues/674#issuecomment-421415815

j4yz, 2019-01-04:
@z3ntu Here's the output of lsusb:
```
Bus 001 Device 003: ID 1532:0037 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0037 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
```
And the ls output concerning the Razer:
```
lrwxrwxrwx. 1 root root   9 Jan  4 21:30 usb-Razer_Razer_DeathAdder_2013-event-if01 -> ../event8
lrwxrwxrwx. 1 root root   9 Jan  4 21:30 usb-Razer_Razer_DeathAdder_2013-event-mouse -> ../event4
lrwxrwxrwx. 1 root root   9 Jan  4 21:30 usb-Razer_Razer_DeathAdder_2013-if01-event-kbd -> ../event5
lrwxrwxrwx. 1 root root   9 Jan  4 21:30 usb-Razer_Razer_DeathAdder_2013-if02-event-kbd -> ../event9
lrwxrwxrwx. 1 root root   9 Jan  4 21:30 usb-Razer_Razer_DeathAdder_2013-mouse -> ../mouse1
```

As I have a Windows partition on my system I could easily provide screenshots from the Synapse software if you need them.

Thanks!