openrazer/openrazer #677: modprobe: ERROR: could not insert 'razerkbd': Operation not permitted

gun1x, 2018-09-13:
I am having troubles modprobing razerkbd:
```
 INSERT  gunix  ~  sudo modprobe -v razerkbd
insmod /lib/modules/4.18.5-200.fc28.x86_64/extra/razerkbd.ko.xz 
modprobe: ERROR: could not insert 'razerkbd': Operation not permitted
```

However dkms seems to work:
```
 INSERT  gunix  ~  sudo dkms install openrazer-driver/2.3.1
Module openrazer-driver/2.3.1 already installed on kernel 4.18.5-200.fc28.x86_64/x86_64
```

Here is a list of installed packages:
```
 INSERT  gunix  ~  rpm -qa | grep -i -e kernel -e razer
kernel-modules-4.18.5-200.fc28.x86_64
kernel-modules-4.17.18-200.fc28.x86_64
openrazer-kernel-modules-dkms-2.3.1-2.1.noarch
openrazer-meta-2.3.1-2.1.noarch
kernel-core-4.17.19-200.fc28.x86_64
kernel-4.18.5-200.fc28.x86_64
kernel-headers-4.18.5-200.fc28.x86_64
kernel-modules-4.17.19-200.fc28.x86_64
kernel-core-4.18.5-200.fc28.x86_64
kernel-tools-4.18.5-200.fc28.x86_64
kernel-devel-4.18.5-200.fc28.x86_64
kernel-4.17.19-200.fc28.x86_64
openrazer-daemon-2.3.1-2.1.noarch
kernel-core-4.17.18-200.fc28.x86_64
kernel-4.17.18-200.fc28.x86_64
kernel-tools-libs-4.18.5-200.fc28.x86_64
python3-openrazer-2.3.1-2.1.noarch
```

z3ntu, 2018-09-13:
It seems like you have Secure Boot enabled. Disable it or sign your DKMS modules.

gun1x, 2018-09-13:
Indeed.

```
 INSERT  gunix  ~  dmesg| grep -i secure
[    0.000000] secureboot: Secure boot enabled
[    0.000000] Kernel is locked down from EFI secure boot; see man kernel_lockdown.7
```

I have some reading up to do, to understand how secure boot works and if I want to disable it or to sign the packages. However this is off-topic for openrazer, so I think the ticket can be closed.