openrazer/openrazer #678: Razer DeathAdder 3500 (2016, 1532:0054) support

GrigoriyMikhalkin, 2018-09-15:
Hi, `Razer DeathAdder 3500 (2016)` mouse is not supported by `openrazer`:

```
$ lsusb -d 1532: -v

Bus 005 Device 002: ID 1532:0054 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0054 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

$ ls -lah /dev/input/by-id/

lrwxrwxrwx 1 root root   9 Sep 15 12:03 usb-Razer_Razer_DeathAdder-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Sep 15 12:03 usb-Razer_Razer_DeathAdder-mouse -> ../mouse0
```

z3ntu, 2018-09-15:
* Please post the output of
`lsusb -d 1532: -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

Synaxis, 2018-09-18:
support for deathadder 2013 (the classic) 6400 dpi

GrigoriyMikhalkin, 2018-11-05:
@z3ntu Hi, i'm trying to add support for this mouse myself, as an exercise. I added new class to `daemon/openrazer_daemon/hardware/mouse.py` which inherits from `__RazerDeviceSpecialBrightnessSuspend`, rebuild package and restarted daemon. It's wrote in log: `Loaded device specification: RazerDeathAdder3500 --------------------- (1532:0054)` but mouse itself wasn't found.

I found, that there is method `daemon/openrazer_daemon/daemon.py::RazerDaemon::_load_devices` which loops through each device in the system and checks if there any hardware class that it belongs to. It checks it with `daemon/openrazer_daemon/hardware/device_base.py::RazerDevice::match` method, which has two conditions:

1. Device should have same VID and PID as hardware class
2. Device path should contain `device_type`

My mouse passes 1st condition, but not 2nd. My mouse path is `/sys/devices/pci0000:00/0000:00:1c.5/0000:06:00.0/usb5/5-2/5-2:1.0/0003:1532:0054.0006` and it doesn't contain `device_type`. Could you explain, please, what is `device_type`, why it's needed and is there any workaround to this problem? OS that i'm using is `Ubuntu 16.04.5 LTS`

z3ntu, 2018-11-06:
You have to add support in the driver for the device first.

GrigoriyMikhalkin, 2018-12-15:
@z3ntu Hi, so i added support in the driver and installed it. But it didn't recognize mouse automatically, i have to manually bind it to razermouse driver. Is that expected behavior?

commit 79bfa565 (2019-02-06): Add support for the Razer DeathAdder 3500

Fixes #678
closed 2019-02-06 by z3ntu