openrazer/openrazer #683: Python API DBus Unable to set arguments

sci4me, 2018-09-24:
I am trying to run the custom_starlight.py example and getting this:

```
Found 1 Razer devices

ERROR:dbus.connection:Unable to set arguments (b'\x00\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',) according to signature None: <class 'ValueError'>: embedded null byte
ERROR:dbus.connection:Unable to set arguments (b'\x00\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',) according to signature None: <class 'ValueError'>: embedded null byte
Traceback (most recent call last):
  File "custom_starlight.py", line 84, in <module>
    device.fx.advanced.draw()
  File "/usr/lib/python3.7/site-packages/openrazer/client/fx.py", line 600, in draw
    self._draw(bytes(self.matrix))
  File "/usr/lib/python3.7/site-packages/openrazer/client/fx.py", line 592, in _draw
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "custom_starlight.py", line 61, in starlight_effect
    device.fx.advanced.draw()
  File "/usr/lib/python3.7/site-packages/openrazer/client/fx.py", line 600, in draw
    self._draw(bytes(self.matrix))
  File "/usr/lib/python3.7/site-packages/openrazer/client/fx.py", line 592, in _draw
    self._lighting_dbus.setKeyRow(ba)
  File "/usr/lib/python3.7/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.7/site-packages/dbus/connection.py", line 641, in call_blocking
    message.append(signature=signature, *args)
ValueError: embedded null byte

    self._lighting_dbus.setKeyRow(ba)
  File "/usr/lib/python3.7/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.7/site-packages/dbus/connection.py", line 641, in call_blocking
    message.append(signature=signature, *args)
ValueError: embedded null byte
```

z3ntu, 2018-09-24:
CC @lezed1

lezed1, 2018-09-24:
@sci4me: I haven't tried this on Python 3.7 yet, that would be my first guess as to the source of the error.

Do other examples work, most importantly the custom zone one which does almost the same thing without threading?

sci4me, 2018-09-24:
@lezed1 No, the custom zone example does the same thing. I do have Python 3.7.0.

z3ntu, 2018-09-24:
Which device do you have? And have you tried rebooting?

sci4me, 2018-09-24:
@z3ntu I have a Razer Blade Stealth (1532:0205). I have rebooted multiple times.

z3ntu, 2018-09-24:
I can reproduce with the fake/emulated device, thanks for the info.

z3ntu, 2018-09-24:
Also technically a duplicate of #350 :wink:

z3ntu, 2018-09-24:
c7eda4d3 broke it btw

closed 2018-09-24 by z3ntu

gmalmquist, 2019-07-11:
Hey @z3ntu ; I appear to be running into this same issue. `custom_starlight.py` and `custom_zones.py` fail with the exact same errors described above. The other scripts work. Net result: using things like `polychromatic`, I'm able to set the color of the whole keyboard or leave it cycling, but can't control individual keys :(

**Hardware**:
Brand new Razer Blade 15 (2018) (https://www.razer.com/gaming-laptops/razer-blade/shop#gtx1060--fhd-60hz-8thgen--256plus-blk) 

**Software**:
I'm running Manjaro, with openrazer installed with pacaur according for the instructions for Arch, with version `openrazer-meta 2.5.0-1`.

`list_devices.py` reports:

```
Found 1 Razer devices

Razer Blade 15 (2018) Base Model:
   type: keyboard
   serial: BY1924A32602222
   firmware version: v1.0
   driver version: 2.5.0
```

An unfounded guess: I wonder if the 2018 models they're selling today are slightly different from the 2018 models they actually sold in 2018?

gmalmquist, 2019-07-11:
As an interesting note, it still fails  in the same way if I comment out the code which actually modifies the matrix:

```
# Set random colors for each zone of each device
for device in devices:
  rows, cols = device.fx.advanced.rows, device.fx.advanced.cols

  #for row in range(rows):
  #  for col in range(cols):
  #    device.fx.advanced.matrix[row, col] = random_color()

  device.fx.advanced.draw()
```

gmalmquist, 2019-07-11:
Actually, the error message isn't exactly identical-- the bytestring in the error message isn't all zeros in my case. It definitely does lead with a null byte though:

```
python3 custom_zones.py 
Found 1 Razer devices

ERROR:dbus.connection:Unable to set arguments (b'\x00\x00\x0f.\x00\xd5\xbf\x00d\x00\x8clR\xff\x00b\x00\xd2"\x00B\xf7\x00$]\xff\x00\xa1\t\x00\x00m\xa0x\x00%\x00@Y\xc1\x00=\x00~8\x00\xa0x\x00Dw\x01\x00\x0fr\xc3\x00\x00\x91v\x00b\xbcn\xd2\x00\x00o\x04k\x00\xde\xa2P\x00~\x9c\x00\xe1\x00r\x007\xbe\x13\xac\x00\x00\x8d\x1e\xc9\x00,|\x00\xb7\x00\xd5[\x06\x00\xca\x02\x00\x0f\x00p(\x0f\x9f\x00|\x7f\x00\x00\x9bN\xc2;\x00:;\x007\x00x\x00\xf6\x80\xf1\x00\x7f\x98\x00K_\xe1\x00\xdb\x00}5\x00\xf6\x00v\xe7\x14\x00lcL\x00\x03\x00\x0f\x19\x00\x19i\xd5\x00q\xfd\x00\x15\xa5\x00\n2\x002R\x00\x00\xf9p#\x8f\x00m\x00\xee\x9d\x00H\x9dD\x00E\x00N8\x004\xb7\x00r-\xf9\x00J\x005\x04\x00\x0fz\x00\x89\x00\xa2o\xd3\x0e\x00,\x00\xbb\x00\x8aN/`\x00\x00ERaF\x005.\x00L\x00\xffS(\x00\x8c\x17\x009b\x00fl\x00\x00\\bx\xdc\x00\x05\x00\x0f\x9b:\x00\x963\x00\x00\x1dl\\\x008\xbd}\x00v\xd4\x00\xd97\x00\x8e\x00O3\x00\xaa\x00\x02\x97{\x84\x00\x10N\x00\x00\xcb-\x00d\xe3i\xe9\x00\x0c\x00\xc3',) according to signature None: <class 'ValueError'>: embedded null byte
Traceback (most recent call last):
  File "custom_zones.py", line 41, in <module>
    device.fx.advanced.draw()
  File "/usr/lib/python3.7/site-packages/openrazer/client/fx.py", line 600, in draw
    self._draw(bytes(self.matrix))
  File "/usr/lib/python3.7/site-packages/openrazer/client/fx.py", line 592, in _draw
    self._lighting_dbus.setKeyRow(ba)
  File "/usr/lib/python3.7/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.7/site-packages/dbus/connection.py", line 641, in call_blocking
    message.append(signature=signature, *args)
ValueError: embedded null byte
```

lezed1, 2019-07-11:
So the matrix operations issues seem to be common in razer blade sleath laptops. I had issues with mine at first until Terry figured out the matrix includes a number pad with isn't on the laptop. I think we thought this was the result of reusing other keyboard led firmware. I wonder if there is a similar issue here?

gmalmquist, 2019-07-11:
I realize that it'll probably be hard to sort out without access to my machine so let me know if there's anything I can do to troubleshoot, and I'll poke at it tomorrow. I'm a software engineer I'm quite familiar with Python, but I don't know y'all's code and drivers are not normally my jam.

z3ntu, 2019-07-11:
@gmalmquist Do you have any message like `Couldn't add method to DBus:` in the daemon log (`~/.local/share/openrazer/logs/razer.log`)?

gmalmquist, 2019-07-11:
Doesn't look like it https://gist.github.com/gmalmquist/cb02a50468da772c059fc690680215c8

commit 78e59de0 (2018-09-24): Fix stupid error and add logging so it gets noticed the next time

Fixes #683, fixes #350