openrazer/openrazer #684: Razer BlackWidow Elite (2018, 1532:0228)

ananace, 2018-09-25:
```
# lsusb -d 1532: -v

Bus 001 Device 020: ID 1532:0228 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0228 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow Elite
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

# ls -lah /dev/input/by-id

lrwxrwxrwx 1 root root   9 Sep 25 18:02 usb-Razer_Razer_BlackWidow_Elite-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 Sep 25 18:02 usb-Razer_Razer_BlackWidow_Elite-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Sep 25 18:02 usb-Razer_Razer_BlackWidow_Elite-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root  10 Sep 25 18:03 usb-Razer_Razer_BlackWidow_Elite-if02-event-mouse -> ../event27
lrwxrwxrwx 1 root root   9 Sep 25 18:02 usb-Razer_Razer_BlackWidow_Elite-if02-mouse -> ../mouse1
```

I'll provide package captures and other information once I've had time to set up a proper VM.

ananace, 2018-09-26:
Images from Razer Synapse 3;

[image]
[image]
[image]
[image]
[image]

And some PCAP files;

[Razer PCAPs.zip](https://github.com/openrazer/openrazer/files/2419572/Razer.PCAPs.zip)

Hopefully that's enough information?

fiercebrake, 2018-11-20:
Greetings!

I just got this keyboard, and would like to request for it to be supported by the openrazer driver.

As per teh instructions, i am attaching the outputs for the related commands.

Let me know if there is anything else to help on this effort!

[blackwidow_elite_0.txt](https://github.com/openrazer/openrazer/files/2601529/blackwidow_elite_0.txt)
[blackwidow_elite_1.txt](https://github.com/openrazer/openrazer/files/2601530/blackwidow_elite_1.txt)

MLDMoritz, 2019-02-02:
Still no way to use this?

thecodemonk, 2019-02-04:
Anything else needed? Would like this working as well and I'm willing to test/provide any information needed.

Naor-Tedgi, 2019-03-01:
any updates for drivers for linux?

DrDerpyMD, 2019-03-05:
Is this device still unsupported?

beranat, 2019-03-30:
Dear Everyones, I have implement [draft support for the keyboard](https://github.com/openrazer/openrazer/pull/804), so it would be very appreciated to check support - more users find more issues.

*NOTE*: Razer Synapse 3 generates some effects manually (look it sends 'set color for every key') and some effects or some its parts may be not implemented by hardware.

Severus157, 2019-04-25:
Any news about when the Support is going to be included in openrazer? Looking for a way to install it, since I got my Blackwidow Elite a few days ago.
If anyone knows how to install the testing version please give me a hint :)

NitroBAY, 2019-04-25:
Follow the wiki instructions here : https://github.com/openrazer/openrazer/wiki/Building-a-package except you change the url from github.com/openrazer/openrazer TO https://github.com/beranat/openrazer

Severus157, 2019-04-26:
Thanks. Took me a while to figure out which PKGBUILD is the correct one, but I think I managed to edit it correctly. At least it installed and Keyboard is recognised perfectly.

SkylerMews, 2019-04-28:
How exactly do I get this to work? I've went through the guide linked above and have it installed. I replaced the git repo with the edited one, but still no dice. Polychromatic still doesn't recognize it ... What am I missing ... ?

MLDMoritz, 2019-04-28:
Please, this ticket is not to support people on how to compile this.

fiercebrake, 2019-05-09:
Hello everyone,

When compiling/installing @beranat branch, my keyboard stops working (1532:0228); getting back to the master branch build solves the problem. Any ideas ?

Atamisk, 2019-05-25:
> Dear Everyones, I have implemented [draft support for the keyboard](https://github.com/openrazer/openrazer/pull/804), so it would be very appreciated to check support - more users find more issues.
> 
> _NOTE_: Razer's Synapse 3 generates all effects manually (look it sends 'set color for every key') and some effects or its parts may be not implemented by hardware.

I am testing your draft support and have a couple of small corrections and issues. Would you consider opening your fork to issues so I can add my notes there? 

(Typed up on my shiny new Blackwidow Elite :D)

beranat, 2019-05-25:
Dear @Atamisk, that's very kind of you. I have sent invitation.

richscze, 2019-06-09:
And idea on when this change can be merged into the master branch and delivered?  I'd like to buy this keyboard but I'm waiting on support.

closed 2019-07-13 by z3ntu

tildearrow, 2019-07-14:
Just to note, since issue actions are too small that they're often skipped: It has been done.

commit 03a48901 (2019-07-13): Add Razer BlackWindow Elite support

Fixes #684, fixes #803