openrazer/openrazer #685: crash on Razer blade stealth with fedora

gedaiu, 2018-09-28:
```
➜  ~ sudo journalctl -b | grep -i razer_mount
Sep 28 11:01:58 bo-sz razer_mount[5900]: Driver razerkbd
Sep 28 11:01:58 bo-sz razer_mount[5901]: Driver razerkbd
Sep 28 11:01:58 bo-sz razer_mount[5902]: Driver razerkbd
Sep 28 11:01:58 bo-sz razer_mount[5915]: Device_ID 0003:1532:0232.0006
Sep 28 11:01:58 bo-sz razer_mount[5905]: Device_ID 0003:1532:0232.0004
Sep 28 11:01:58 bo-sz razer_mount[5916]: Modprobing razerkbd
Sep 28 11:01:58 bo-sz razer_mount[5917]: Modprobing razerkbd
Sep 28 11:01:58 bo-sz systemd-udevd[5853]: Process 'razer_mount razerkbd 0003:1532:0232.0004' failed with exit code 1.
Sep 28 11:01:58 bo-sz systemd-udevd[5860]: Process 'razer_mount razerkbd 0003:1532:0232.0006' failed with exit code 1.
Sep 28 11:01:58 bo-sz razer_mount[5907]: Device_ID 0003:1532:0232.0005
Sep 28 11:01:58 bo-sz razer_mount[5926]: Modprobing razerkbd
Sep 28 11:01:58 bo-sz systemd-udevd[5858]: Process 'razer_mount razerkbd 0003:1532:0232.0005' failed with exit code 1.
```
```
➜  ~ tail ~/.local/share/openrazer/logs/razer.log
2018-09-28 11:04:58 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2018-09-28 11:04:58 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2018-09-28 11:04:58 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2018-09-28 11:04:58 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2018-09-28 11:04:58 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2018-09-28 11:04:58 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2018-09-28 11:04:58 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2018-09-28 11:04:58 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2018-09-28 11:04:58 | razer                          | INFO     | Serving DBus
2018-09-28 11:12:32 | razer                          | DEBUG    | DBus called get_serial_list
```

z3ntu, 2018-09-28:
What's the output of `sudo dkms install openrazer-driver/2.3.1` and `sudo modprobe razerkbd` ?

gedaiu, 2018-09-28:
```sudo dkms install openrazer-driver/2.3.1
[sudo] password for gedaiu: 

Creating symlink /var/lib/dkms/openrazer-driver/2.3.1/source ->
                 /usr/src/openrazer-driver-2.3.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
KERNELDIR=/lib/modules/4.18.10-300.fc29.x86_64/build make driver....
cleaning build area...

DKMS: build completed.

razerkbd.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.18.10-300.fc29.x86_64/extra/

razermouse.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.18.10-300.fc29.x86_64/extra/

razerfirefly.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.18.10-300.fc29.x86_64/extra/

razerkraken.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.18.10-300.fc29.x86_64/extra/

razermug.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.18.10-300.fc29.x86_64/extra/

razercore.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.18.10-300.fc29.x86_64/extra/
Adding any weak-modules

depmod....

DKMS: install completed.
```

```
sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Operation not permitted
```

z3ntu, 2018-09-28:
Disable Secure Boot or sign your dkms modules.

closed 2018-09-28 by z3ntu

gedaiu, 2018-09-28:
@z3ntu thanks!