openrazer/openrazer #689: How About Naga Molten?

jeferod83, 2018-10-01:
I have a Razer Naga Molten Special Edition mouse, model number RZ01-0028.

lsusb says:

Bus 001 Device 003: ID 1532:0015 Razer USA, Ltd Naga Mouse

This device are not covered by the driver. Is there a way to include it? Thanks!

z3ntu, 2018-10-02:
* Please post the output of
`lsusb -d 1532: -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

jeferod83, 2018-10-03:
Hi @z3ntu !

Here is the outputs:

`lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:0015 Razer USA, Ltd Naga Mouse
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0015 Naga Mouse
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      47
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1

ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 220 out  2 20:47 .
drwxr-xr-x 4 root root 580 out  2 20:47 ..
lrwxrwxrwx 1 root root   9 out  2 20:47 usb-046d_0825_7C629990-event-if00 -> ../event8
lrwxrwxrwx 1 root root   9 out  2 20:47 usb-0d8c_USB_Sound_Device-event-if03 -> ../event7
lrwxrwxrwx 1 root root  10 out  2 20:47 usb-©Microsoft_Xbox_360_Wireless_Receiver_for_Windows_FF416EF0-event-joystick -> ../event16
lrwxrwxrwx 1 root root   6 out  2 20:47 usb-©Microsoft_Xbox_360_Wireless_Receiver_for_Windows_FF416EF0-joystick -> ../js0
lrwxrwxrwx 1 root root   9 out  2 20:47 usb-Razer_Razer_Naga-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 out  2 20:47 usb-Razer_Razer_Naga-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 out  2 20:47 usb-Razer_Razer_Naga-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 out  2 20:47 usb-Thermaltake_Tt_eSPORTS_KNUCKER_Gaming_Keyboard-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 out  2 20:47 usb-Thermaltake_Tt_eSPORTS_KNUCKER_Gaming_Keyboard-if01-event-kbd -> ../event4`

Model No: RZ01-0028, Laser Gaming Mouse
Product No: RZ01-00280500

The SS's, I'll take it soon.

z3ntu, 2018-10-04:
This seems to be exactly the same as the "Razer Naga Classic". So I'll close this as a duplicate of #139. In that issues, screenshots and packet captures are still missing though.

closed 2018-10-04 by z3ntu

jeferod83, 2018-10-04:
I believe that this is true. The request #139 is freeze. I just take a few screenshots, and pcap I'll send it later. Can you reopen my request? I just wanna help Openrazer to rise. Thanks a lot!

z3ntu, 2018-10-05:
Please move further discussion to #139