openrazer/openrazer #693: Mamba Elite Support

aneergaard, 2018-10-11:
I just purchased a Mamba Elite, any work in progress to make this compatible? I'm happy to submit whatever in order to facilitate the process, when I get the mouse.

aneergaard, 2018-10-16:
Razer Mamba Elite
Product No: RZ01-02560100
Model No: RZ01-0256

Output of `lsusb -d 1532: -v`:

```Bus 001 Device 004: ID 1532:0232 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0232 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
[... 169 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

Output from `ls -lah /dev/input/by-id/`:
```
alexno@alexno-rbs:~$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 240 Oct 16 09:06 .
drwxr-xr-x 4 root root 600 Oct 16 09:06 ..
lrwxrwxrwx 1 root root  10 Oct 14 06:33 usb-11918920-0081IA59_USB_Camera-event-if00 -> ../event17
lrwxrwxrwx 1 root root   9 Oct 14 06:33 usb-ELAN_Touchscreen-event-if00 -> ../event8
lrwxrwxrwx 1 root root   9 Oct 14 06:33 usb-Razer_Razer_Blade_Stealth-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Oct 14 06:33 usb-Razer_Razer_Blade_Stealth-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Oct 14 06:33 usb-Razer_Razer_Blade_Stealth-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Oct 14 06:33 usb-Razer_Razer_Blade_Stealth-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Oct 16 09:06 usb-Razer_Razer_Mamba_Elite-event-mouse -> ../event18
lrwxrwxrwx 1 root root  10 Oct 16 09:06 usb-Razer_Razer_Mamba_Elite-if01-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10 Oct 16 09:06 usb-Razer_Razer_Mamba_Elite-if02-event-kbd -> ../event20
lrwxrwxrwx 1 root root   9 Oct 16 09:06 usb-Razer_Razer_Mamba_Elite-mouse -> ../mouse3
```

I'll follow up on the packet capture and screenshots.

aneergaard, 2018-10-16:
Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

terrycain, 2018-10-29:
> As I know, the only difference between Mamba Elite and Mamba Tournament
> Edition is in the sensor. Elite has optical and TE has laser. So I think
> that enabling Mamba Elite shouldn't be problematic at all. I'll try to pull
> this repo and modify it to use the same code for Elite and if it will be
> working well, I can create PR for this change.

👍

IvanVojtko, 2018-10-30:
So far my progress is, that I have fully working logo and scroll wheel. I hope that I'll solve this today.

VilNeo, 2018-11-06:
Hello,
I purchased a Mamba Elite as well.
Above lsusb command just results in product id 0x006c.
Ivan, did you have further progress?

Best regards
Alex

IvanVojtko, 2018-11-08:
So I've managed to enable this mouse on linux, but you can only change colors and brightness of wheel and logo. But syncing with my keyboard works fine, so I have enabled spectrum effect and synced keyboard with mouse using RazerGenie. There is a problem, that this project doesn't support new way of setting colors/effects which came with Synapse V3. If you want, I can try to create pull request, just to have this mouse properly recognized and to be able to sync with other Razer devices.

VilNeo, 2018-11-08:
A pull request would be nice.
Thanks for the effort.

VilNeo, 2018-11-09:
I managed to find the packet which sets specific lights on the Mamba Elite:
001f 0000 0041 0f03 0000 0000 1300 0010 0000 2000 0030 0000 4000 0050 0000 6000 0070 0000 8000 0090 0000 a000 00b0 0000 c000 00d0 0000 e000 00f0 0010 0000 1010 0010 2000 1030 0010 4000 0000 0000 0000 0000 0000 0000 0000 0e00
Here is the mapping:
Byte 1-5: No clue. Maybe subcommand and fading speed?
Byte 6-8: Wheel color
Byte 9-11: Logo color
Byte 12-38: Left side color from front to back
Byte 39-65: Right side color from front to back

IvanVojtko, 2018-11-12:
Ok so I've managed to fully control colors and brightness, but there is a big BUT. New devices relay entirely on "daemon". You need to remember last colors, you need to spam colors to spectrum mode, you need to spam brightness values to use breath effect...For example if you want to restore brightness from 0 to 255, you need to tell which color you want. Not just brightness. It is a lot of work. I want just tweak color changing and brightness setting and I think that's all. I don't even think that they will merge my PR, because it is a mess.

IvanVojtko, 2018-11-13:
Ok so I've created initial PR. You can play with it and test it if you like. Also if anyone is using Arch linux, I can provide you with custom PKGBUILD to create package with my PR. #712

sleef54, 2019-01-11:
I played with the PR commits, and seems to render my mouse inoperable actually...(using make install)

Ability to allow the mouse to suspend when locking the screen would be ideal minimally.

Running Fedora 29; happy to help.

Judgement84, 2019-01-12:
I'm on Solus and also have this mouse. I can send any logs needed to help facilitate getting this working. (Would love to have it sync with my keyboard and mousepad)

IvanVojtko, 2019-01-14:
Guys I've managed to have it "synced" with other things like keyboard and mousepad, also mouse is turning off when I lock the screen, but it's a bit "hacky".

Judgement84, 2019-01-14:
Does the sync allow it to pick up the pattern that the keyboard is doing? (i.e. Can it do waves?)

Smarre, 2019-01-29:
Has there been work to get this properly implemented? I can look if I can help if there’s no progress.

samindaperamuna, 2019-03-09:
Any official support for this fork yet?

wschlich, 2019-03-28:
I also just got a Mambe Elite and I'm willing to help. What needs to be done?

z3ntu, 2019-03-28:
Please get packet captures of the device when using Synapse (basically going through the options in Synapse and recording the USB traffic), see https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Smarre, 2019-04-05:
@z3ntu: is there any specific information what should be captured from mice? All buttons clicked? Something specific about the Synapse software? That page just talks about some ”static mode”, what is that?

z3ntu, 2019-04-07:
@Smarre It's mostly just clicking through the options in Synapse, you can leave out key rebinding and macro stuff as that's just in software

Smarre, 2019-04-07:
@z3ntu Okay I’ll see what I can get for you.

Smarre, 2019-04-07:
@z3ntu I tried to collect everything. 

[razer_mamba_elite.zip](https://github.com/openrazer/openrazer/files/3051687/razer_mamba_elite.zip)

Please say if I missed something that’s useful.

samindaperamuna, 2019-05-04:
Any update on this?

DrDustwolf, 2019-05-09:
I've got my new Mamba Elite today. And I'm looking for Linux Support with Openrazer too. How far is the progress with this mice?

samindaperamuna, 2019-06-11:
Why is this closed again?

spezifant, 2019-06-12:
@samindaperamuna it's not closed. The referenced duplicate #856 got closed.

Just got a ME as well. If I can provide anything let me know.

St-14, 2019-06-22:
Any traction on this?  Let me know if I can help.

samindaperamuna, 2019-08-03:
Is anyone working on this?

mlouage, 2019-08-11:
I'm also the happy owner of a Mamba Elite. Any update on this?

IvanVojtko, 2019-08-29:
Hey guys. For anyone using arch linux, try to build my package.
https://drive.google.com/file/d/1uESlierpfuZNc8g5pW8hoPVmWzVNu_-e/view?usp=sharing

Build, install, enable systemd service, maybe reboot. Then first run this command:
`echo -n -e "\x00\x02" > /sys/bus/hid/drivers/razermouse/*/device_mode`
This should turn "custom" mode on your mouse and enables you to push custom colors or so.
This command:
`echo -n -e "\xf5\x79\x00" > /sys/bus/hid/drivers/razermouse/*/matrix_effect_static`
should change color of your mouse. Those are RGB hex values. Also razer genie should recognize your mouse and allows you to change DPI, polling rate, brightness and color. Also you can turn on sync and you should have the same color for all devices. Also mouse should turn off/on based on lock screen event. Let me know if it works.

nolanlikeeh, 2019-09-08:
Hi Ivan,
I have a Mamba Elite and tried your work so far. Thanks so much!

> Build, install, enable systemd service, maybe reboot. Then first run this command:
> `echo -n -e "\x00\x02" > /sys/bus/hid/drivers/razermouse/*/device_mode`
> This should turn "custom" mode on your mouse and enables you to push custom colors or so.

This command gave my mouse a rainbow effect similar to all pictures of the mouse you see on the internet although it was static. Still much better in my opinion than default when bought.

> This command:
> `echo -n -e "\xf5\x79\x00" > /sys/bus/hid/drivers/razermouse/*/matrix_effect_static`
> should change color of your mouse. Those are RGB hex values. 

This gave my mouse a static cream colour.

>Also razer genie should recognize your mouse and allows you to change DPI, polling rate, >brightness and color. Also you can turn on sync and you should have the same color for all >devices. Also mouse should turn off/on based on lock screen event. Let me know if it works.

Razer Genie did recognise my mouse and I could change the colour/DPI but when I changed lighting to 'none' I can't get the colour back and so now my mouse has no colour. 
I have tried the initial command to try and get back the rainbow effect but no command so far returns lighting to my mouse.

Thanks for any help, and again your work so far.

IvanVojtko, 2019-09-08:
Hi. Just turn on brightness all the way up :) None effect means, that it will dim the mouse to zero. You can do it through razer genie or using command line:
`echo -n "255" > /sys/bus/hid/drivers/razermouse/*/matrix_brightness`
I am not sure, if 100 is maximum or 255 but it should work anyway :)

nolanlikeeh, 2019-09-22:
Thanks, needed the command.

> Hi. Just turn on brightness all the way up :) None effect means, that it will dim the mouse to zero. You can do it through razer genie or using command line:
echo -n "255" > /sys/bus/hid/drivers/razermouse/*/matrix_brightness

samindaperamuna, 2019-11-07:
Any progress on this driver?

beranat, 2019-12-01:
Dear All, 
Please check the extended support [Mamba Elite v1.1 effect support](https://github.com/openrazer/openrazer/pull/960):
also update [Polychromatic](https://github.com/beranat/polychromatic) - patched version supports the mouse.

z3ntu, 2020-03-29:
Device is supported now.

closed 2020-03-29 by z3ntu

samindaperamuna, 2020-03-29:
This doesnt seem to work with Polychromatic daily builds.

z3ntu, 2020-03-29:
@samindaperamuna It will be included in the next release. In the meantime on Ubuntu you can use the daily ppa or on Arch Linux use `openrazer-git`. The next release should be done in the next couple of weeks.

samindaperamuna, 2020-03-30:
Im using RazerGenie on Debian and it's working perfect.
Thanks.