openrazer/openrazer #694: Razer Blackwidow Stealth

nikospaz, 2018-10-12:
It stopped working and the keyboard is recognized as BlackWidow Ultimate

z3ntu, 2018-10-12:
You have the device 1532:011B ?

nikospaz, 2018-10-12:
No, I think is the 0214

Στάλθηκε από το iPhone μου

z3ntu, 2018-10-13:
Can you make a photo of the bottom of your keyboard with the label?

nikospaz, 2018-10-13:
[image1.jpeg]

Στάλθηκε από το iPhone μου

z3ntu, 2018-10-13:
It literally says `[image1.jpeg]`, I guess GitHub doesn't forward image attachments. You can attach the image directly on GitHub or upload it to some image hoster and paste the URL here.

nikospaz, 2018-10-13:
https://ibb.co/fqtDtp

Στάλθηκε από το iPhone μου

z3ntu, 2018-10-13:
Well it seems that the Ultimate and the Ultimate Stealth are the same internally. I don't see what should be wrong here.

closed 2018-10-13 by z3ntu

nikospaz, 2018-10-13:
Thank you very much I will uninstall and reinstall everything and I will tell you!!

Στάλθηκε από το iPhone μου