openrazer/openrazer #695: Razer Blade 15 support

nakrule, 2018-10-14:
Would be create to have the new Razer laptop to be supported.

z3ntu, 2018-10-14:
`lsusb | grep 1532` output please

enrapture6, 2018-10-14:
I got a 2018 razer blade 15 I am trying to running arch on. 

I get lockup crash to an ext4 fs error when waking from lid close suspend/hibernate. Other times bootup, reboot, lag.

I had the same issues when running ubuntu. Switch to arch as i though controlling all the packages would avoid issues. My conclusion is that even the latest kernel does not support the new hardware yet. 

[enrapture6@razer ~]$ lsusb | grep 1532
Bus 001 Device 004: ID 1532:0233 Razer USA, Ltd 
[enrapture6@razer ~]$

If there are anymore out that one would like to see, let me know.

nakrule, 2018-10-16:
$ lsusb | grep 1532
    Bus 001 Device 004: ID 1532:0233 Razer USA, Ltd

z3ntu, 2018-10-18:
Duplicate of #605 (in short: it's already supported, but not yet in a release - which I'll create soon)

closed 2018-10-18 by z3ntu