openrazer/openrazer #696: Support for Razerblade 2018

kawmaiparis, 2018-10-20:
In the openrazer web, the Razerblade 2018 is listed as a supported device but it is not listed in the polychromatic daemon log.

lah7, 2018-10-20:
It could be possible OpenRazer was improperly installed, so the daemon won't recognise the device. [Have a look at this troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting) in case it's due to missing kernel headers;  or if your user isn't part of the `plugdev` group or if it's because secure boot is enabled.

If none of those troubleshooting suggestions work and you've tried rebooting, let us know what you've tried and your distro.

lah7, 2018-10-20:
Oops, turns out the Blade 2018 hasn't hit **stable** yet (thanks for alerting me @z3ntu).

There should be a new release in the next day or two. In the meantime, install the `daily` packages for your distribution if there's one available. In Arch, it's `openrazer-meta-git` or in Ubuntu, the daily PPA:

    sudo add-apt-repository ppa:openrazer/daily
    sudo apt update
    sudo apt install --upgrade openrazer-meta

z3ntu, 2018-10-21:
Duplicate of #605

closed 2018-10-21 by z3ntu