openrazer/openrazer #697: Daemon not responding.

Smesh292, 2018-10-20:
marina@marina-Aspire-E5-571:~$ openrazer-daemon -Fv
Starting daemon.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 125, in run_daemon
    test_dir=args.test_dir)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 100, in __init__
    self._device_classes = openrazer_daemon.hardware.get_device_classes()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/__init__.py", line 24, in get_device_classes
    imported_module = __import__(hw_module, globals=globals(), locals=locals(), fromlist=['*'], level=0)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/mouse.py", line 13, in <module>
    from openrazer_daemon.misc.key_event_management import NagaHexV2KeyManager as _NagaHexV2KeyManager
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 175
    usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py    def _poll_read(self):
                                                                                 ^
SyntaxError: invalid syntax

Stopping daemon.

marina@marina-Aspire-E5-571:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful

lah7, 2018-10-20:
Just to check, which version of `python3-daemonize` and `python3` do you have installed?

    apt-cache show python3-daemonize | grep Version
    python3 --version

The `daemonize` version in the Ubuntu archives is a bit old, so the PPA contains a newer version, which should be `2.4.7`. If it's not, you can download its [deb here](https://launchpad.net/~openrazer/+archive/ubuntu/stable/+files/python3-daemonize_2.4.7-artful_all.deb) from the [stable PPA](https://launchpad.net/~openrazer/+archive/ubuntu/stable/+packages).

Smesh292, 2018-10-21:
marina@marina-Aspire-E5-571:~$ apt-cache show python3-daemonize | grep Version
Version: 2.4.7-artful
Version: 2.3.1-1
marina@marina-Aspire-E5-571:~$ python3 --version
Python 3.6.3
marina@marina-Aspire-E5-571:~$

lah7, 2018-10-21:
Unfortunately, I'm unable to reproduce the error (OpenRazer 2.3.1) - it's working fine on a 17.10 installation. Ubuntu 17.10 also reached end of life.

I'd suggest reinstalling OpenRazer and all of its dependencies since there shouldn't be a syntax error. Otherwise, try the daily builds from 18.04 (bionic) by adding `ppa:openrazer/daily` and edit the following source file to point to `bionic` instead of `artful`:

    /etc/apt/sources.list.d/openrazer-ubuntu-daily-artful.list

Followed by `sudo apt update` and `sudo apt install openrazer-meta`, the latest version at this time is 2.4.0.

closed 2018-10-21 by lah7