openrazer/openrazer #698: Razer mamba 2012 : Battery level not consistent after two reads

dlezcano, 2018-10-20:
I noticed if I read the charge_level or charge_status, I have alternatively 0 and something around 200 (which I assume it is correct).

cd /sys/devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.3/3-1.3:1.0/0003:1532:0025.000E

for i in $(seq 1 10); do sudo cat charge_level; sleep 1; done
234
0
232
0
232
0
234
0
232
0

If I don't put a delay between the reads, the output is:

for i in $(seq 1 10); do sudo cat charge_level; done
232
0
0
0
0
0
0
0
237
0

And sometimes dmesg shows:

[1204397.839685] razer driver: Device data transfer failed.
[1204397.840772] razer driver: Invalid USB repsonse. USB Report length: -32
[1204397.840777] razermouse: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
[1204398.607741] razer driver: Device data transfer failed.
[1204398.608856] razer driver: Invalid USB repsonse. USB Report length: -32
[1204398.608859] razermouse: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
[1204405.408440] razer driver: Device data transfer failed.
[1204405.409626] razer driver: Invalid USB repsonse. USB Report length: -32
[1204405.409630] razermouse: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
[1204406.321157] razer driver: Device data transfer failed.
[1204406.322300] razer driver: Invalid USB repsonse. USB Report length: -32
[1204406.322305] razermouse: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
[1204407.073304] razer driver: Device data transfer failed.
[1204407.074287] razer driver: Invalid USB repsonse. USB Report length: -32
[1204407.074307] razermouse: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .

At this point, charge_level will return always zero until I reset the device.

z3ntu, 2018-10-20:
Yeah we know. Currently nothing we can really do about it unfortunately.

dlezcano, 2018-10-21:
Is the root cause known ?

z3ntu, 2018-10-21:
Yes, the devices sometimes don't respond with a proper response, but with a "busy" or whatever error which isn't handled.

dlezcano, 2018-10-21:
ok, thanks.
Any plan to upstream the driver ?

dlezcano, 2018-10-30:
Is there any plan to upstream the razer driver ?

z3ntu, 2018-10-31:
Not really. There are some other issues in this repo about upstreaming the kernel drivers with more details.

closed 2019-02-28 by z3ntu