openrazer/openrazer #699: Deathstalker Essential 1532:006e

dastious, 2018-10-21:
Please add support for Deathstalker Essential 1532:006e.
Would help as much as possible.

Model #: RZ01-0254
Product #: RZ01-02540100

Screenshot : https://imgur.com/a/jJ4upHf

**lsusb -d 1532: -v**

```
Bus 003 Device 002: ID 1532:006e Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x006e 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

**ls -lah /dev/input/by-id/ | grep Razer**
```
lrwxrwxrwx 1 root root   9 21.10.2018 12:58 usb-Razer_Razer_DeathAdder_Essential-event-mouse -> ../event0
lrwxrwxrwx 1 root root   9 21.10.2018 12:58 usb-Razer_Razer_DeathAdder_Essential-if01-event-kbd -> ../event1
lrwxrwxrwx 1 root root   9 21.10.2018 12:58 usb-Razer_Razer_DeathAdder_Essential-if02-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 21.10.2018 12:58 usb-Razer_Razer_DeathAdder_Essential-mouse -> ../mouse0

```

dastious, 2018-10-21:
Here some pcap for :

Changing brightness to 33 to 100 and go back, and move the bar.
Switching profile sensitivity and enter some dpi and switch.
Switching between Static and breathing light effect

[pcap.zip](https://github.com/openrazer/openrazer/files/2499312/pcap.zip)

Here another pcap for moving this bar : https://imgur.com/a/W2MSfgU
[sensi_stage_off_movebar.zip](https://github.com/openrazer/openrazer/files/2499314/sensi_stage_off_movebar.zip)

z3ntu, 2018-10-21:
Could you please save the packet captures as `.pcapng` and not `.pcap` (you'll probably have to redo them 
- if it takes much time, just do one again, save as pcapng again and send it and I'll take a look at it if it's properly done)

dastious, 2018-10-21:
I did it with https://desowin.org/usbpcap/ from windows, it doesn't seem to support pcapng. Is pcapng is just an extension or something more special ? Maybe i need to redo them in vm ?

It seem there is tools to convert pcap to pcapng but i'm not sure if it is useful for you.

z3ntu, 2018-10-21:
So did you use Wireshark or really just usbpcap? With Wireshark .pcapng should be the default - and no, at least with Wireshark the conversion from .pcap to .pcapng doesn't yield the same results as just saving it as .pcapng

dastious, 2018-10-21:
just usbpcap

dastious, 2018-10-21:
Ok founded how to usbmon directly in wireshark on windows.
Here is the log in pcapng format. Don't hesitate to ask more pcapng info or anything. Thanks!
[pcapng.zip](https://github.com/openrazer/openrazer/files/2499401/pcapng.zip)

dastious, 2018-12-13:
Is it working ?

aschi2403, 2019-02-05:
I'm curious, are you still working on support for the DeathAdder Essential? Is there a possibility I can contribute to that?

dastious, 2019-02-06:
dunno about @z3ntu but if you know how to extract data from the pcap, i think i can update the code.

guglovich, 2019-03-04:
Should we wait for support?

dastious, 2019-03-05:
Try to wireshark it to see if we can see the protocol. Maybe my wireshark was not good.

Aagame3, 2019-03-20:
Is this still being continued? Just got this mouse and would like to configure it.

halfelf, 2019-04-15:
There is another version of DeathAdder Essential, the "White Edition". Seems identical with the black one except the color.  Please add support for it too.

Model #: RZ01-0254  
Product #: RZ01-**02540200**   

`lsusb -d 1532: -v`

```
Bus 001 Device 004: ID 1532:0071 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0071 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

`ls -lah /dev/input/by-id/ | grep Razer`

```
lrwxrwxrwx     9 root 15 Apr 15:01  usb-Razer_Razer_DeathAdder_Essential_White_Edition-event-mouse -> ../event6
lrwxrwxrwx     9 root 15 Apr 14:48  usb-Razer_Razer_DeathAdder_Essential_White_Edition-if01-event-kbd -> ../event7
lrwxrwxrwx     9 root 15 Apr 14:48  usb-Razer_Razer_DeathAdder_Essential_White_Edition-if02-event-kbd -> ../event8
lrwxrwxrwx     9 root 15 Apr 15:01  usb-Razer_Razer_DeathAdder_Essential_White_Edition-mouse -> ../mouse0
```

dastious, 2019-04-18:
make a pcap guys ...

lsdage, 2019-06-12:
> make a pcap guys ...

Hello,

what are the requirements on the pcap?
What environment? Linux? Windows? Any software needed other than Wireshark?
Any special settings for Wireshark?

Thanks!

dastious, 2019-06-13:
there is a wiki and readme https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

lsdage, 2019-06-19:
> make a pcap guys ...

Gave it a shot... 
[razerDeathAdderEssential-pcapng.zip](https://github.com/openrazer/openrazer/files/3306430/razerDeathAdderEssential-pcapng.zip)

lsdage, 2019-06-28:
Hi,

after a few trials and errors I got the DeathAdder Essential recognized by the driver.

Setting the DPI seems to work ok (I am using RazerGenie).

commit c2f1d00c (2019-07-01): Add Razer DeathAdder Essential initial support (#869)

Fixes #699
closed 2019-07-01 by z3ntu