openrazer/openrazer #700: razerkbd: Response doesnt match request

parkerlreed, 2018-10-23:
Arch
openrazer-daemon-git 2.4.0.r0.g2aa1558-1
openrazer-driver-dkms-git 2.4.0.r0.g2aa1558-1
openrazer-meta-git 2.4.0.r0.g2aa1558-1

Stealth Late 2016
````
Bus 001 Device 006: ID 1532:0220 Razer USA, Ltd Blade Stealth (2016)
````

dmesg
````
[   94.299280] razerkbd: Response doesnt match request. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  101.738959] razerkbd: Response doesnt match request. Start Marker: 02 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  108.172460] razerkbd: Response doesnt match request. Start Marker: 02 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000ff00ff00000000ffff .
[  108.373147] razerkbd: Response doesnt match request. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f00ffff000000000000ffff00 .
[  134.717539] razerkbd: Response doesnt match request. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  147.584547] razerkbd: Response doesnt match request. Start Marker: 02 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  159.447906] razerkbd: Response doesnt match request. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  169.300225] razerkbd: Response doesnt match request. Start Marker: 02 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  177.543536] razerkbd: Response doesnt match request. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  180.157712] razerkbd: Response doesnt match request. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  182.770558] razerkbd: Response doesnt match request. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  185.184278] razerkbd: Response doesnt match request. Start Marker: 02 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
[  195.841463] razerkbd: Response doesnt match request. Start Marker: 02 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05000f000000000000000000000000 .
````

z3ntu, 2018-10-23:
Do you notice any actual issues or do you just see this in the dmesg?

parkerlreed, 2018-10-23:
Just see that in dmesg. Setting effects and everything works fine.

z3ntu, 2018-10-24:
Then it's fine.

closed 2018-10-24 by z3ntu