openrazer/openrazer #701: Openrazer unable to find Razer Blade Stealth (Early 2017)

gragundier, 2018-10-24:
The daemon works fine.  Even though I'm on arch linux, I didn't seem to have rebuild the packages.
However, when I try `python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`
I get output: []
So, the problem is laptop isn't being detected.  The part I am new to and need help with is on how to fix this problem or determine if I can fix it at all.

z3ntu, 2018-10-24:
Does `sudo modprobe razercore` output anything?

gragundier, 2018-10-24:
`modprobe: FATAL: Module razercore not found in directory /lib/modules/4.18.14-arch1-1-ARCH`
and thank you in advance.

z3ntu, 2018-10-25:
`sudo dkms install openrazer-driver/2.4.0` ?

gragundier, 2018-10-25:
`[sudo] password for gragundier: 

Creating symlink /var/lib/dkms/openrazer-driver/2.4.0/source ->
                 /usr/src/openrazer-driver-2.4.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.18.14-arch1-1-ARCH cannot be found at
/usr/lib/modules/4.18.14-arch1-1-ARCH/build or /usr/lib/modules/4.18.14-arch1-1-ARCH/source.`

z3ntu, 2018-10-25:
Read the message and install your kernel headers :)

gragundier, 2018-10-26:
OHhhhhhhhh thx.

closed 2018-10-26 by gragundier